necroware/gameport-adapter #1: digital mode switching necessary?

git-moritz, 2021-05-27:
as far as I understand, we don't need to initialize the Sidewinder 3D Pro into "digital transfer mode" at all... every "trigger" causes the joystick to transmit a 64-bit-package, regardless of mode. the only advantage of digital mode is that IF we use both trigger lines (X1 and Y2), the joystick will transfer triplets instead of bits, which allows for transmission error correction.

since we are only using single-line-transfer, is there any reason why we even _try_ to switch into digital mode?

necroware, 2021-05-27:
Hi Moritz and thank you for the donation first of all :)

We need to switch into digital mode even when using only single line transfer. In analog mode joystick will not react on any trigger signals and will instead emulate CH FlighStick or Thrustmaster analogue joystick, dependent on the current joystick setting.  You can just use the 3D Pro in CH FlightStick mode as well, it is already a feature rich configuration, but in that case you would only have 4 buttons and the hat switch would only work in 4 ways. To use the full features set of the 3D Pro, you need to set it into digital mode.

git-moritz, 2021-05-27:
that's actually not true :)
I used parts of your code to make a sketch that only uses readPacket (via readState) without ever calling enableDigitalMode, and it works just as well.
This behavior is in fact documented in US Patent # 5628686A:

**"The first interrupt request is interpreted by the digital joystick 102 as the Send Data command. In response to the Send Data command, the digital joystick 102 transmits the 64-bit data packet with digital position data and button data, as described above."**

(the sequence that switches the stick into digital mode consist of four interrupt requests - the first of these is already a "Send Data" command!)

**"A single mode data bit is also included in the 64-bit data packet. The mode data bit indicates whether the digital joystick 102 is operating in the Digital Transmission Mode 300 (see FIG. 4) or in the Analog Emulation Mode 302 as well as the type of conventional joystick 10 being emulated if in the Analog Emulation Mode."**

The 64-bit-packet can be transmitted in ANY mode, and even _tells_ you the mode it's currently in. (We could, in fact, even use the mode switch as an additional input "button"...)

necroware, 2021-05-27:
Which joystick did you use to test it? At least on my 3D Pro it doesn't work. The initial state is analog mode and you can try to trigger it as much as you want, you'll get no response. I analyzed it with logic analyzer and without the "magic sequence" it doesn't do anything. I'll get two more Sidewinder joysticks in the next days, one is again a 3D Pro the other is Precision Pro , and see how they react. I have here a Sidewinder GamePad as well and that one is initially in digital mode already and works the way you describe, but not the 3D Pro. A side note, the SW GamePad has 15 bits packages and not 64. Also the GP can be set into information state as well and deliver 72 bit  packages with firmware version info and such things. I don't use it currently.

git-moritz, 2021-05-27:
I am using my Sidewinder 3D Pro. The position of the analog emulation switch does not appear to matter.
As far as I understand, the "full" digital mode was supposed to be used for possible future versions of sidewinder devices, which could be combined: one device acts as a "hub" and relays all data to the PC - additional devices are connected to the first device. For that, a higher transfer rate would have been necessary, which is why they used all four button lines.

git-moritz, 2021-05-27:
...aaaaand of course I made a stupid mistake and did not realize that the "readState" function calls "enableDigitalMode" via "reset".

so I guess you are right, the sequence is needed after all. funny, because the patent says it is not. hmm.

necroware, 2021-05-27:
Yes, the adapter makes auto reset, in case the read fails for some reasons. Very similar approach is made in Linux kernel module as well. I noticed read errors from time to time on my test setup, so I also implemented an auto-reset in case of on-going problems.

In regards of the patent, it is just a hint and not necessarily a document, where it is defined how the whole thing exactly has to work. So we can't completely rely on every statement in there. In the end all the Sidewinders are working similar, but in details differently though.

git-moritz, 2021-05-27:
I suppose I'll close this issue, then :)

closed 2021-05-27 by git-moritz