necroware/gameport-adapter #9: Logitech Cyberman 2 doesn't work

necroware, 2021-11-09:
The device senses correctly all the axes (6 axes in total, 3 translate + 3 rotate) and all the 8 buttons, but then... No reaction at all to the movement of any axis nor to the press of any button!

necroware, 2021-11-09:
Well, unfortunately I have no Cyberman 2 here to investigate, so you will probably have to try to find out what is wrong on your own. I just can tell, that if the Joystick is recognized properly, then the adapter can read the packages from the joystick properly. So the issue must be somewhere in the state decoder. If really nothing moves, not one button works and no other things are happening, like for example you press a button and an axis is jumping, or other way around you move and axis and some buttons are jumping, then the decoder seems to stop the operation very early.

If you are in C/C++, I'd start with the check of the packet size. See Logitech.cpp:63. You can use Serial.print() to send some values onto the serial console and see in the Arduino IDE Serial Monitor what comes out. May be this way you can spot the line, where it fails. As I said, it's going to be very complicated for me to make a remote assumption. From the Logitech devices I have only the Wingman Extreme Digital and that's the only one I tested so far.

necroware, 2021-11-09:
Answered by Shores on [Vogons](https://www.vogons.org/viewtopic.php?p=1016709#p1016709):

Hi!

So, I've fiddled a bit with the sketch, and added a serial dump of the received packet bits to the readPacket() function.
This shows that the readPacket function does not always read the packet correctly from the CyberMan 2, since it mostly returns packets made of only 1, 2 or 3 bits.

If in the update() function I put

```
if (packet.length != m_metaData.packageSize) {
  enableDigitalMode();
  const auto discardMetaPacket = readPacket();
  return false;
}
```

which resets the digital mode, discards the metadata packet and then waits for the next update, then the first two or three packets after that do come in correctly (64bit packets), and then it goes back to 1/2/3 bit packets, and gets reset again, and so on...

In this strange way the adapter works properly, and in the joystick control panel I can see all the buttons and axes working properly, albeit a little slow!

Any hint on how can I proceed, what to check/test/change ?

Thanks!

necroware, 2021-11-09:
Ok, that is a step forward, but is not a solution yet beaming face with smiling eyes May be we read the packets too fast, I observed similar problem for the Sidewinder before and added a cooldown function there. You can try to insert a delayMicroseconds(2000) into the beginning of the update() function. Right into the first line. This will enforce the adapter to wait 2ms before every read. Let's see if it changes anything. You also can play with the value and increase/decrease it.

ShoresOfNowhere, 2021-11-15:
Hi, adding a 

delay(5u); 

as the first line of update() in Logitech.h makes all work properly in terms of reading values from the device, but something is still not working properly, and that is the match between the first 3 axes (which are correctly mapped to x,y and z respectively) and the next 3 axes which represent rotation on each axis...

Windows sees axis 4 as rotation on x, 5 as on y and 6 as on z, but by moving the controller I can see this is not true: axis 4 is rotation on z, 5 on x and 6 on y.

What is interesting is that upon detection of the device, it responds with a device name of ZXY! I think this is the device telling the system how the axes 4 to 6 should be mapped to the first three axes!

I don't know if this means that different Cyberman2 copies return this differently, so the device name should be parsed and axes mapped differently at each boot, or all the devices do the same mapping...

I'll try and write this mapping code, but i'll be slow in that since i broke my left wrist and so I'm single handed in writing code for a month!

necroware, 2021-11-15:
Hi,

very nice, but delay of 5ms is IMHO too much.  We should try to keep the input lag as low as possible. Try to get that number down as much as possible. May be you can get it even under 1ms using delayMicroseconds(...).

In regards of axis mapping, don't worry I'm working on that, since Cyberman 2 is not the only one device, which has now a wrong mapping. It actually doesn't make any difference, since it's just a cosmetic thing, because you can simply remap the axes as you like. But for the explanation, I made a huge refactoring to simplify the code and to make it easier to add new devices, so currently the USB code is unified and during the registration it doesn't know which axis has which function. So it just numerates them as they come from the joystick. Apparently in the order x,y,z,rx,ry,rz and this is what you see. I have an idea how to improve that in the future, but this is out of scope of this issue. But you can summarize here, how the axes should be mapped for this joystick.

ShoresOfNowhere, 2021-11-15:
I must correct myself: the device reports itself as ZYX but the axes 4 to 6 should be mapped as rZ, rX and rY , so the device name has no meaning as to that.

A for the delay, if you look at 

https://elixir.bootlin.com/linux/latest/source/drivers/input/joystick/adi.c#L73

which is the working ADI protocol driver for linux, you see at row 497 that the polling interval is set to 20 milliseconds, and on row 35 you see that the delay after a data packet is set to 4 milliseconds, so I don't think it would be right to lower it so much.

In fact, by testing, my cyberman II does NOT work with a 2 milliseconds (or 2000 microseconds) as you told me to test the code; by testing, the minimum value to make it work properly is approx

delayMicroseconds(3750u);

Anything below that would simply make the code not work...

ShoresOfNowhere, 2021-11-15:
As I said the minimum working delay is 3.75 milliseconds or 3750 microseconds; under that it no longer works.

By the way, no pressure at all, instead thanks for all your hard work on this project!

necroware, 2021-11-15:
Thank you too for testing! I thinkg 3750 or 5000 doesn't make a huge difference anymore :) So if 5ms works, it's fine.
I just tested the code with the Wingman Extreme Digital and pushed the code to github. You can checkout the latest main commit.

https://github.com/necroware/gameport-adapter/commit/adee2a455a4b91ffadd3c862859fd118833132ab

If you confirm, that it works, I'll close this issue.

ShoresOfNowhere, 2021-11-16:
Confirming it just works properly! Should we add CyberMan2 to the compatibility on the main readme?

necroware, 2021-11-16:
Thank you for the feedback! Yes, I'll add the CyberMan 2 to the list.

closed 2021-12-14 by necroware

crashdot, 2022-09-25:
Hi,
I have tried two new Necroware adapters and two Cyberman 2 controllers and can't get them to work together. What comes up is a Sparkfun Pro Micro 2-axis, 2-button controller that doesn't respond. If the problem may be old firmware, can anyone direct me where I can learn how to update the Necroware firmware? I see the files listed here but don't know what to do with them.
Thanks

crashdot, 2022-09-25:
Somebody sent me basic instructions to get me started:

The software is programmed using the Arduino IDE software that can be downloaded here
[Software | Arduino](https://www.arduino.cc/en/software)

Download the gameport firmware here

[codeload.github.com](https://codeload.github.com/necroware/gameport-adapter/zip/refs/tags/v0.2.1)
[//codeload.github.com/necroware/gameport-adapter/zip/refs/tags/v0.2.1](https://codeload.github.com/necroware/gameport-adapter/zip/refs/tags/v0.2.1)

Connect the gameport connector USB to the PC with the appropriate cable

If you dont have the pro micro as a board option you might need to download the sparkfun boards addon first or configure IDE to find the board information.

Go to the Preferences (File > Preferences). Then, towards the bottom of the window, paste this URL into the "Additional Board Manager URLs" text box:

`https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json`

Under Tools Board manager, set the board as Pro Micro and select the COM port it is attached to

Under Tools, select Processor as 16mhz ( important! )

Open the .ino file in the gameport firmware folder with the arduino IDE.

Select the Upload arrow button and wait until it completes.

necroware, 2022-09-25:
Cyberman 2 uses ADI protocol, so you need to set the switches to 1001. However, there was a bug in the firmware, which sometimes could lead to wrong switches detection. To update the firmware, just download the Arduino IDE and open `gameport-adapter.ino` file there. Then select the `Arduino Micro` device in the menu, connect your adapter to USB and press `Upload` in the IDE. I see, that you use `Sparkfun Pro` and not the `Arduino Micro`, but I think, they are the same. Otherwise just read how to use the Arduino IDE with the `Sparkfun Pro`. Google is your friend.

crashdot, 2022-09-25:
Sorry, I have never used an Arduino before and did not have the IDE software. Some of the documentation I did find described the steps for installing the Arduino drivers in Windows, although I have not been able to find it in the device manager, determine the device name, or determine the COM port, since gaming controllers do not seem to show up unless you select "View devices by container" which I have done maybe once in my life before today. I did not know this was a Sparkfun board and was just searching for "Pro Micro" since that is what is printed on the board. I Googled "upgrade necroware gameport adapter firmware" but did not find any procedure. I did need a few more pieces of the puzzle to get there. Thanks very much for your help! This thing is great :)

necroware, 2022-09-25:
I just made a new release of the firmware with all the fixes, which landed in the firmware so far. Please use this for your experiments, or take the main branch, if you always want to test the bleeding edge. The lates release can be always found here: https://github.com/necroware/gameport-adapter/releases