necroware/gameport-adapter #10: Support for multiple daisy-chained SideWinder devices

afmenez, 2021-12-14:
The [SideWinder Gamepad](https://en.wikipedia.org/wiki/Microsoft_SideWinder#Gamepad) had a pass-through port, so additional gamepads or joysticks could be used without unplugging the SideWinder, and also allowed the connection of up to four SideWinder gamepads working simultaneously.

hadess, 2021-12-22:
@necroware let me know if you want a spare Microsoft Sidewindow Game Pad 1.0 (90873) with passthrough. I just bought one, and didn't realise it was digital only and wouldn't work with older DOS games.

necroware, 2021-12-22:
@hadess thank you very much, but since @afmenez opened this ticket, I think, that he wants to provide an implementation. Currently unfortunately I'm short on time, so I can't take care of this project as well. Furthermore I decided for myself not to implement any daisy chaining for any joysticks (f.e. Gravis supports it too). Back in the days there was usually only one GamePort on the PC, that's why some joystick manufacturers came up with the daisy-chaining eventually. With USB today there is actually no real need for that, since you usually have a lot of USB ports and you simply can take one adapter for each joystick separately. I'm working on many projects in parallel and have to decide what is really important and what is just nice to have. This feature is in my opinion just a nice to have one, so I see it with a very low priority. In other words, I probably will not implement it any time soon. But I'd be glad if @afmenez would like to contribute an implementation. I'll definitely will take time to review and integrate it if it comes.

hadess, 2021-12-22:
> With USB today there is actually no real need for that

Makes sense

afmenez, 2021-12-22:
I have several SideWinder devices, including 2 GamePads.  But you need only one to implement daisy-chaining, since you can connect any other SideWinder to the GamePad.
I am not sure if I agree with the multiple USB ports argument, since I would need two adapters to connect two SideWinder devices.  Would this work?

afmenez, 2021-12-22:
I would be willing to take a shot implementing this, but currently I have no access to the adapter.  Also, I am a software guy and the parts are hard to come by here in Brazil.  Is there a way to buy one already complete?

necroware, 2021-12-22:
> I would need two adapters to connect two SideWinder devices. Would this work?

Yes, you can use one adapter for every joystick and connect as many of them as you want. You would need to have as many adapters as joysticks, which you want to connect, but they cost next to nothing if you make them yourself.

> I would be willing to take a shot implementing this, but currently I have no access to the adapter. Also, I am a software guy and the parts are hard to come by here in Brazil. Is there a way to buy one already complete?

I'm also just a software engineer. Unfortunately I don't sell the adapters, since I'd have to pay a lot of money for license to be able to sell such parts in the EU. You can order all the parts from China (I do as well), also the PCBs

dferyance, 2022-01-17:
I may (time permitting) be able to help contribute to this. I have two sidewinder gamepads. I'm probably more knowledgeable with reverse-engineering protocols than dealing with USB support -- which I expect USB would have to be modified to support two or more joysticks. 

@afmenez regarding limited adapter access, here are some ideas. You don't need the dip switch to use the adapter, you can simply hard-code the interface to use. You probably also could skip the PCB and wire the Arduino into a gameport or even wire it to a joystick pins. I believe the Arduino Lenoardo would be compatible as it has the same chip if that helps any. But for sure, a PCB + pro micro + gameport header is going to be simpler than all of that. Just suggesting options. I used different parts than the digikey numbers as I had similar but different parts on hand already. So there are options.

JocPro, 2022-02-10:
Hi there.

I'm willing to help in any way I can. I ordered some PCBs and components to build a couple of adapters (I hope they arrive in a month or so) but I think I can build one with some spare parts that I have lying around in a protoboard. I also happen to own four working Sidewinder Gamepads and a retro PC with a working gameport. I have basic electronics and programming knowledge.

sonik-br, 2022-10-04:
I don't have the adapter (yet) and not a single sidewinder gamepad.
But I can help with the USB HID stuff. @dferyance please tag me if you start to work on this.

dferyance, 2024-01-22:
I just submitted a PR for daisy chaining. I limited it to 2 just because I only have 2 gamepads to test with. Anyone have 4? If you want to try 4 update the max joysticks and the max gamepads constants and the code and I think it should work.

I didn't address passthrough mode. My suggestion is to split that request into its own issue as that is a different kind of feature.

@sonik-br

Rikki-Tikki-Tavi-12, 2024-04-10:
I have a suggestion to make, and I understand if some may not immediately find it appealing, but hear me out.
What if, in the unlikely case that a fourth sidewinder gamepad is connected, we don't register it as a fourth device, but instead register the third controller to have 14 extra buttons. Even in case there is no way to use these buttons in the application as-is, they could be remapped to a device in the os level. I believe steam comes with with a tool to just that, for example.

I understand that this is a bit of a dirty solution, that may not appeal to everyone. There should definitely be a code switch to deregister the com port, to enable the fourth controller, but I think everyone agrees that the project shouldn't be shipped like this by default, whereas I see no issue shipping it with the behavior described above by default. Most users will be entirely unaffected, and only in the edge case of someone assembling four functional artifacts of ancient origin, they would still be able to use them with minor inconvenience.

What do you guys think?

dferyance, 2024-04-10:
I had initially considered an idea similar to this. Although I was thinking of packing all 4 into the same joystick. There are two downsides I know of. Multiplayer games expect each player to have their own joystick and don't generally allow splitting a joystick between users. You mention re-mapping tools. I've never used them. It isn't something I considered. It might work but for sure that is an added complication.

The other one has to do with the USB descriptor. In early versions of my code, I allocated a descriptor per joystick. I started seeing some odd problems possibly due to the increased memory usage. It is possible the problems were caused by something else but memory usage is important. We'd have to allocate 2 descriptors in order to have the two mappings. That probably will be safe but memory is at a premium. It would also complicate the code having to manage the two descriptors.

So yeah, I like the serial port disable solution that I put in. That's what I'm going to run. I wish the bootloader was easier to work with. You have to reset into bootloader mode by shorting two pins. Then in arduino set your serial port. Then as you build and upload, you have to reset it again twice. Once so that arduino knows that port is available and another time right when it is done compiling so it can upload. This is very finicky and timing sensitive.

I looked at the bootloader firmware. It appears the only way to enter the bootloader mode is via the reset pin instead of a software reset and there is no way to keep it in bootloader mode more than a few seconds. I had hoped we could use a dip switch setting to enter bootloader mode... but not without PCB changes. At one time I was uploading on command-line via avr-dude and had better luck but I don't think this is a reasonable expectation for people less technical or not wanting to learn all the technical details.

And the arduino USB code doesn't allow for turning on and off the USB serial on-the-fly. Or else I'd want serial enabled for any mode other than sidewinder or enabled if you have fewer than 4 joysticks connected. But I didn't see a way without re-writing the arduino USB code.

So that is the rational behind this solution. I wanted 4 to work because I like vintage hardware and think it is neat to use all the features. But practically this isn't so important as you can always run multiple joystick adapters.

Rikki-Tikki-Tavi-12, 2024-04-10:
I think it's quite possible you are the last person on planet Earth playing (I presume) Micro Machines with six Sidewinder game pads, so if you have no appetite for this, then I guess, I will shelve the idea until annother weirdo (no offense) comes along. :)

Cheers.

MacGH23, 2024-04-10:
Since I only use one Adapter for the 4 Pads, it is OK for me to use the reset Button to update the FW ;-)