necroware/gameport-adapter #11: Support for the SideWinder Freestyle Pro

afmenez, 2021-12-14:
The [SideWinder Freestyle Pro](https://en.wikipedia.org/wiki/Microsoft_SideWinder#Freestyle_Pro) was a gamepad with motion-sensing control.

afmenez, 2021-12-22:
I have this device and I would be willing to take a shot implementing this, but currently I have no access to the adapter.

timixretroplays, 2022-12-22:
I also have this gamepad, and (medium-term) am looking into making a bunch of these adapters to help make them more accessible generally. I'd be willing to work towards distributing them to people who can contribute code for new controllers (this level of reverse-engineering is not in my wheelhouse, but basic Arduino/PCB stuff is).

In the worst case - @afmenez , would it be at all useful for me to provide you with testing/debug/logging info, using my own adapter and a Sidewinder? I could run/test whatever code you send me.

YARD2, 2023-02-27:
The Freestyle Pro is already USB ready and was sold with an adapter
https://www.adminxp.com/hardware/index.php?aid=116

closed 2023-02-27 by necroware