necroware/gameport-adapter #12: Support for the SideWinder Force Feedback Pro

afmenez, 2021-12-14:
The [SideWinder Force Feedback Pro](https://en.wikipedia.org/wiki/Microsoft_SideWinder#Force_Feedback_Pro) was like a Precision Pro with motors for force feedback effects.

afmenez, 2021-12-22:
I have this device and I would be willing to take a shot implementing this, but currently I have no access to the adapter.

No0ne, 2022-02-02:
Hi! I have a SideWinder Force Feedback Wheel and I just tested sniffing the MIDI output from my Win98 PC while the original driver is running.
[image]
Maybe this works by setting up a USB MIDI interface within the adapter and just forwarding those SysEx and other messages? But I think the hardware needs to be modified so that the TX of the onboard UART is connected to the MIDI out pin.

necroware, 2022-02-02:
That's awesome! Thank you very much. I will take a closer look at it on the weekend. So far I don't use the RX/TX pins on purpose, because I like to keep the serial  port free for debugging. That makes the life easier during the development. I usually do this in all of my Arduino projects and use the RX/TX only if I need a really fast serial connection.... I'll have to think about it.

But your data is already very exciting and gives a lot of stuff to think about ;)

No0ne, 2022-02-02:
Cool! I've another issue, how to debug this?
[image]
I'm just slowly turning the wheel and pressing the pedals but there seems some bits to be shifted around?

caradas, 2022-02-02:
I tried to implement daisychaining on the SW Gamepad last weekend and had something looking almost exactly the same.
The problem was that the 3rd bit of the 3-bit wide bus was read to late. 
To confirm it i changed the order of lines 176-178 on Sidewinder.h to move the problem arround by reading b1 or b2 last.

No0ne, 2022-02-03:
Thanks, but I tried every of the six permutations, didn't make any difference 🤔

No0ne, 2022-02-03:
Ah, just discovered the serial debug feature:
```
Trying to enable digital mode
Guessing model by packet size of 0
Trying to enable digital mode
Guessing model by packet size of 0
Trying to enable digital mode
Guessing model by packet size of 0
Trying to enable digital mode
Guessing model by packet size of 11
Detected model 4
Detected device: MS ForceFeedBack Wheel
Packet decoding failed 1 time(s)
Packet decoding failed 2 time(s)
Packet decoding failed 3 time(s)
Packet decoding failed 4 time(s)
Packet decoding failed 5 time(s)
Packet decoding failed 6 time(s)
Sidewinder init...
Guessing model by packet size of 11
Detected model 4
Packet decoding failed 7 time(s)
Sidewinder init...
Guessing model by packet size of 11
Detected model 4
Packet decoding failed 8 time(s)
Sidewinder init...
Guessing model by packet size of 11
Detected model 4
Packet decoding failed 1 time(s)
Packet decoding failed 2 time(s)
Packet decoding failed 3 time(s)
Packet decoding failed 1 time(s)
Packet decoding failed 2 time(s)
Packet decoding failed 1 time(s)
Packet decoding failed 2 time(s)
Packet decoding failed 3 time(s)
Packet decoding failed 4 time(s)
Packet decoding failed 1 time(s)
Packet decoding failed 1 time(s)
Packet decoding failed 2 time(s)
Packet decoding failed 1 time(s)
Packet decoding failed 2 time(s)
Packet decoding failed 3 time(s)
Packet decoding failed 4 time(s)
Packet decoding failed 5 time(s)
Packet decoding failed 6 time(s)
Sidewinder init...
Guessing model by packet size of 11
Detected model 4
Packet decoding failed 1 time(s)
Packet decoding failed 1 time(s)
Packet decoding failed 2 time(s)
Packet decoding failed 1 time(s)
Packet decoding failed 2 time(s)
Packet decoding failed 3 time(s)
```

necroware, 2022-02-03:
I opened a new issue about the timing issues to keep this one about FFB Pro

opensourcerebel, 2022-03-13:
@No0ne I am trying to implement FFB for the sidewinder wheel - do you have that sniff in a text format? Or any other resources for that manner?

No0ne, 2022-03-13:
Shure, I will try to connect my mt32-pi and just save the midi output coming in via gpio to a text file and upload it here.

opensourcerebel, 2022-03-13:
@No0ne no need, in this wiki there is enough data: https://code.google.com/archive/p/adapt-ffb-joy/wikis/SidewinderFFBWheel.wiki

Hedda, 2022-07-03:
Any news on Sidewinder FFB Pro support?

necroware, 2022-07-03:
> Any news on Sidewinder FFB Pro support?

Unfortunately no, I have no FFB device and nobody contributed to the implementation so far. However, I know, that couple of people are trying to implement this.

Hedda, 2022-08-04:
@necroware Do you live in Europe and would be willinng to look at implementing support for SideWinder Force Feedback Pro?

My OCD made me collect a few over the years and I might be willing to contribute one if add FFB GamePort Adapter compatibility.

PS: I remember reading someone released a SideWinder FFB firmware for Teensy but not sure if they ever made it open source(?):

https://www.descentbb.net/viewtopic.php?t=19061&start=280

https://code.google.com/archive/p/adapt-ffb-joy/downloads

https://www.reddit.com/r/hotas/comments/qd3kow/ms_sidewinder_ffb_pro_teensy_and_now_it_works_on/

https://github.com/tloimu/adapt-ffb-joy

https://github.com/tloimu/adapt-ffb-joy/wiki

PPS: I by the way also read that Immersion Corp's Force Feedback patents have expired since a few years back now:

https://www.reddit.com/r/hotas/comments/bmnr8f/as_of_today_immersion_corps_force_feedback/

https://patents.google.com/patent/US6275213

https://patents.google.com/patent/US6424333

CyberLuke-GBG, 2022-08-06:
Hi, I'm another software dev and I have this Sidewinder Forcefeedback Pro (USB version). I have a small experience with Arduino and MIDI. 

Currently I bought a great joystick InterAct FX Racing Wheel (SV-281) and it does have force feedback and gameport connector. It works in the same manner as this Microsoft one. It requires sound card with midi (mpu-401) interface. Gameport features both midi in, midi out and serial connection. So it sends MIDI directly to soundcard via gameport and the software driver in the OS will use it for force feedback. It is quite simple. You just need to parse MIDI. The hard part is how to implement USB HID with force feedback for DirectX input?

I definitely need to write something for this InterAct. It's for RC cars, Trackmania, etc. And the issue is it works only in Win98 due to 16-bit VxD driver. There is no WDM driver for Windows 2000/XP. 

I just need to buy some Teensy or Arduino first.

I think the hardest part here will be modifying Arduino USB HID library to output force feedback data and make it work with DirectX input layer.

cyberluke, 2022-08-06:
Hi, I'm another software dev and I have this Sidewinder Forcefeedback Pro (USB version). I have a small experience with Arduino and MIDI.

Currently I bought a great joystick InterAct FX Racing Wheel (SV-281) and it does have force feedback and gameport connector. It works in the same manner as this Microsoft one. It requires sound card with midi (mpu-401) interface. Gameport features both midi in, midi out and serial connection. So it sends MIDI directly to soundcard via gameport and the software driver in the OS will use it for force feedback. It is quite simple. You just need to parse MIDI. The hard part is how to implement USB HID with force feedback for DirectX input?

I definitely need to write something for this InterAct. It's for RC cars, Trackmania, etc. And the issue is it works only in Win98 due to 16-bit VxD driver. There is no WDM driver for Windows 2000/XP.

I just need to buy some Teensy or Arduino first.

I think the hardest part here will be modifying Arduino USB HID library to output force feedback data and make it work with DirectX input layer.

necroware, 2022-08-06:
Hi guys, I also got InterAct FX Racing Wheel (SV-281) some time ago and already was trying to reverse engineer the protocol. I made additional pluggable part, which can be inserted between the joystick and the gameport to grab the communication with a logic analyzer. So it is in the works, but I currently don't know yet how much time it will take to develop everything completely. The SV-281 currently doesn't work at all with this adapter, so the whole protocol has to be reverse engineered. Unfortunately I didn't find any protocol documentation so far.

Hedda, 2022-08-07:
> @necroware Do you live in the EU and are willing to look at implementing support for MS SideWinder Force Feedback Pro?
> 
> My OCD made me collect a few over the years and I might be willing to donate one if it adds FFB GamePort Adapter compatibility.

@necroware Maybe it would make more sense for me to donate/contribute a similar sum to the shipping cost (around $30/€30/£30) to you via PayPal so that you can buy a Microsoft SideWinder Force Feedback Pro joystick from eBay yourself? 

I might perhaps consider donating more if you be willing to send me a few of your Necroware GamePort Adapters in return.

PS: By the way, I live in Sweden myself.

necroware, 2022-08-07:
@Hedda no, you don't need to. I'm currently short on time anyway and I don't think, that I will find any time this year to continue with this project. On my youtube channel I mentioned already couple of times why this year is quite shitty and it currently doesn't look like it's getting better. May be in that time someone else will come up with a solution. There are already couple of people, who wanted to implement FFB.

cyberluke, 2022-08-21:
@necroware Where to get the pluggable part for  SV-281? I can look into that.

lucasstinis, 2023-01-25:
Hi, I'm thinking of getting/building 2 of these adapters, and was wondering if you expect the current 2.1 version of the adapter to be compatible on a **hardware** level with the SW FFB Pro, if support ever gets implemented, or if you think it will require an entirely new board to work properly. Thanks!

necroware, 2023-01-25:
FFB need MIDI pins. They are wired already, so it should be only a question of software, I guess. I would like not to change the compatibility if possible, so everybody can use the same software.

Hedda, 2023-01-26:
> Hi, I'm thinking of getting/building 2 of these adapters, and was wondering if you expect the current 2.1 version of the adapter to be compatible on a **hardware** level with the SW FFB Pro, if [...]

Hardware-wise the original MS Sidewinder FFB Pro should really work like a non-FFB joystick via any Gameport MIDI adapter, the problem is that the force feedback does not work with the software support for it inside the gameport adapter firmware, without custom force feedback software support for the original MS Sidewinder FFB Pro it works just like any dumb non-FFB joystick :(

I am guessing that what necroware means is that while software support can be solved s you should be able to use the same hardware you might end up still having to use a different software (firmware image) for the original MS Sidewinder FFB Pro than for the other joysticks?

PinkBaron1949, 2023-01-29:
Because I still have a Sidewinder Force Feedback Pro and was able to adapt it to MSFS 2020 and link nearly every button to its supposed function, its very interesting what is developing here. I am running an interface ?? that I found on the net.
So where do I get the layout of the pc board and the sofware ? I could not find it on your page. 
regards 
Hans

Hedda, 2023-01-29:
> So where do I get the layout of the pc board and the sofware ? I could not find it on your page.

The PCB board layout and the firmware is in this Git repository on GitHub -> https://github.com/necroware/gameport-adapter

Kicad format project is used for the board -> https://github.com/necroware/gameport-adapter/tree/main/kicad

Arduino IDE to compile source code-> https://github.com/necroware/gameport-adapter/tree/main/firmware/gameport-adapter

PinkBaron1949, 2023-01-29:
Thank you very much, but that is much to technical for me.

lucasstinis, 2023-01-29:
You can always buy a pre-built one from e.g. [here](https://www.tindie.com/products/retrofletch/pc-usb-gameport-adapter-sidewinder-retro-joystick/). Good luck and let's end this convo here as it's not the place for these types of conversations.

Hedda, 2023-01-30:
It is then relatively easy for most hobby/amateur-tinkers to solder the parts by hand as it only has very few simple component, see:

https://www.youtube.com/watch?v=HUmbvfqP784&ab_channel=chadaustinak

## Ordering prototype PCB and assembling yourself

As this is [open-source hardware](https://en.wikipedia.org/wiki/Open-source_hardware) you should know as an option that it is today relatively easy (and inexpensive) to just submit the KiCad PCB layout files directly to some PCB-prototype board manufacturers like Oshpark or PCBWay who will take it then make + ship a few of the PCB boards to you, all without you having to use KiCad yourself or even having to understand the PCB layout:

- https://oshpark.com
   - See example shared project (rev 0.2.1 of the board) -> https://oshpark.com/shared_projects/rzwJjMoP
- https://www.pcbway.com
- https://jlcpcb.com

FYI, Chinese PCB manufacturers will usually make around 20 simple PCB board like this for less than $10 and ship them to you for around $20, so depending on your country's import tax you have those at home for around $30-40 in around a months time.

E.i. just buy the ones from the project BOM (Bill-Of-Matierials) list

https://github.com/necroware/gameport-adapter/blob/main/README.md#bill-of-materials-bom

Note! It is not recommended to buy the Mini-USB variant of Arduino Pro Micro as its header is wider than USB-C and Micro-USB variants), and it is not recommended to buy the Micro-USB variant of Arduino Pro Micro as it is too easy to break the connector.

## Bill of materials (BOM)

The hardware is super simple. To build an adapter you'll need the PCB from this project and following parts:

Part    |  Qty  | [LCSC](https://lcsc.com/) #  | [Digikey](https://www.digikey.com/) #                | [Mouser Electronics](https://www.mouser.com/) # | Comment
--------|-------|---------|--------------------------|----------------------|------------------------------------------
CONN1   |   1   | C77835  | 609-5371-ND              | 523-L77SDA15SA4CH4F  | DB15 female connector
R1..R4  |   4   | C172965 | 13-MFR-25FTE52-100KCT-ND | 603-MFR-25FTE52-100K | 100 kOhm resistors
SW1     |   1   | C15781  | 2449-KG04ET-ND           | 642-DS04T            | DIP-4 switch
U1      |   1   | C72120  | ED3051-5-ND              | 649-DILB24P-223TLF   | DIP24 Socket (optional)
U1      |   1   |   N/A   | 1568-1060-ND             | 474-DEV-12640        | Arduino Pro Micro (ATmega32U4 16MHz, 5V), including two 12 pin header connectors, MicroUSB version (see "Known issues")

Tip! Some other user recopmmended to buy enclosure case: Hammond Manufacturing 1551RTBU, Digi-Key Part # HM1142-ND.

## Firmware

Once you have it and want to flash the firmware yourself (to ensure you have the latest version), you can do that yourself as well.

1. Download source code files from -> https://github.com/necroware/gameport-adapter/tree/main/firmware/gameport-adapter
2. Download the Arduino IDE from -> https://www.arduino.cc/en/software
3. Connect the Adapter with your USB cable.
4. Open "gameport-adapter.ino" included in the source files.
5. In "Tools" -> Port -> select the virtual COM port that says (Arduino Micro).
6. In "Sketch" -> select "Upload" which will compile the firmware and upload it to the Arduino.
7. You are done.

Hedda, 2023-01-30:
I guess that you guys already seen this old Arduino library for the Sidewinder Force Feedback Pro based on the Adapt-FFB-Joy project?

https://code.google.com/archive/p/sidewinder-arduino/

Adapt-FFB-Joy (originally for Teensy 2.0) look to support support Microsoft Sidewinder Force Feedback Pro joistick with force feedback using a seperate firmware image build:

https://github.com/tloimu/adapt-ffb-joy

Project wiki:

https://github.com/tloimu/adapt-ffb-joy/blob/wiki/README.md

Even more interestininly he has in that wiki done his best to documented the MS Sidewinder force feedback MIDI protocol as it has been reverse engineered

https://github.com/tloimu/adapt-ffb-joy/blob/wiki/SidewinderFFBMIDI.md

USB HID descriptor with force feedback development page:

https://github.com/tloimu/adapt-ffb-joy/blob/wiki/USBFfbHid.md

Kreeblah recently also look to have made a custom board for that project based on an ATmega32u4RC-MU to substituted for the original ATmega32u4-MU:

https://github.com/Kreeblah/SidewinderToUSBV2

http://descentbb.net/viewtopic.php?t=15526

Hedda, 2023-01-30:
Cool, also a seperate stand-alone Force Feedback Tester -> https://github.com/tloimu/adapt-ffb-joy/tree/master/FeedbackTester

NolanNicholson, 2023-01-31:
@Hedda - Thanks for collecting all this information in one place. It's really useful!

The 2.1 hardware design for gameport-adapter has a pin hooked up for MIDI TX. It differs from adapt-ffb-joy in two ways.

One, it's on an "ordinary" digital pin, instead of the 32u4's TX pin, so it cannot send MIDI comms using the hardware serial functionality. Arduino has a SoftwareSerial library that would allow sending 31 kbaud messages on pins other than the hardware TX pin, but I'm nervous the overhead of running SoftwareSerial might interfere with performance-sensitive packet decoding. (I tried including it in the firmware, and packets started coming in incorrectly, but I haven't done a ton of testing yet.) It might also be possible to roll something from scratch; MIDI is slow enough.

Two, adapt-ffb-joy puts a 220 ohm resistor in series with the output, which I understand to be pretty typical for MIDI lines. gameport-adapter lacks this resistor. I'm not sure if it's necessary for the Sidewinder.

Hedda, 2023-01-31:
> I believe that the most interesting part is that tloimu has in his wiki for that project as he has there done his best to document the MS Sidewinder Force feedback MIDI protocol as it has been reverse-engineered:
> 
> https://github.com/tloimu/adapt-ffb-joy/blob/wiki/SidewinderFFBMIDI.md
> 
> USB HID descriptor with force feedback development page:
> 
> https://github.com/tloimu/adapt-ffb-joy/blob/wiki/USBFfbHid.md

Another reference is this old but relevant community discussion about reverse engineering the Force Feedback Pro protocol:

https://www.descentbb.net/viewtopic.php?t=19061&start=280

Hedda, 2023-01-31:
> [tloimu](https://github.com/tloimu) made a stand-alone Force Feedback Tester -> https://github.com/tloimu/adapt-ffb-joy/tree/master/FeedbackTester

@NolanNicholson Force feedback for Linux is btw documented (inc. list of force feedback commands under section 5.3.1) here:

https://www.kernel.org/doc/html/v4.17/input/ff.html

FYI, Dirks Software also has a small free Force Feedback Diagnostic Program made for quickly testing MS FFB2 on Windows:

https://www.fs-force.com/support.php -> https://www.dirks-software.ca/downloads/ForceTest.exe

Another tip for testing is to download the free-to-play flight simulation platform/game "DCS World" ("Digital Combat Simulator World", a.k.a "DCS") and test it with (TF-51 Mustang and Su-25T planes are free to play as a demo in the one terrain/map that is included for free), and you should also be able to test it with any helicopters in DCS World (though only the [AH-6 Little bird is available for free as a demo](https://steamcommunity.com/linkfilter/?url=https://forums.eagle.ru/topic/228394-helicopter-efm-demo/)). There are also several free mods from the community that add planes like the A-4 and the T-45. Many people have otherwise recommended the paid-for [Black Shark 3 / Ka-50](https://www.digitalcombatsimulator.com/en/shop/modules/black_shark_3/) DLC for its force feedback experience (as well as other paid-for DLC terrains/maps). Enabled force feedback features depends on the plane/helicopter.

* https://en.wikipedia.org/wiki/Digital_Combat_Simulator
* https://forum.dcs.world/topic/118085-force-feedback/
* https://forum.dcs.world/topic/272322-my-experience-with-force-feedback-in-dcs/
* https://forum.dcs.world/topic/242485-what-should-i-be-getting-out-of-force-feedback/
* https://forum.dcs.world/topic/198916-how-does-dcs-talk-to-force-feedback-joysticks/

> Checkout fork example by JayBee which include an alternative patch set for adaptffbjoy-r54:
> 
> https://github.com/JayBee-git/adapt-ffb-joy

By the way, some more digging shows that JayBee's fork at https://github.com/JayBee-git/adapt-ffb-joy/ contains the latest code of Adapt-FFB-Joy and consolidates fixes and additional features from other forks of adapt-ffb-joy have branches with additional features. That above-mentioned [old also community discussion](https://www.descentbb.net/viewtopic.php?t=19061&start=280) mentions that JayBee's fork included changes from https://github.com/DEV-OGRE/adapt-ffb-joy/tree/feature/shift-and-frame-buffer as well as https://github.com/r8dhex/adapt-ffb-joy/tree/feature/shiftEnable and https://github.com/r8dhex/adapt-ffb-joy/tree/feature/moreButtons).

Hedda, 2023-02-02:
> Cool, [tloimu](https://github.com/tloimu) made a stand-alone Force Feedback Tester -> https://github.com/tloimu/adapt-ffb-joy/tree/master/FeedbackTester

> FYI, Dirks Software also has a small free Force Feedback Diagnostic Program made for quickly testing MS FFB2 on Windows:
> 
> https://www.fs-force.com/support.php -> https://www.dirks-software.ca/downloads/ForceTest.exe

Also check out this project meant to bring Force Feedback functionality to Elite Dangerous but could probably be useful for more:

https://github.com/bobhelander/EDForceFeedback 

fedit.exe under /EDForceFeedback/FFUtils/ is btw apparently Microsoft Force Editor that was included with DirectX back in the day. 

https://documentation.help/Force-Editor/fedit_7xdg.htm

There also looks to be some other related Force Feedback unitilities software there too:

https://github.com/bobhelander/EDForceFeedback/tree/master/EDForceFeedback/FFUtils

Hedda, 2023-02-28:
Any updates on your Microsoft Sidewinder Force Feedback Pro support for Necroware's GamePort Adapter?

Noticed that initial commits for MS FFB Pro without force-feedback support were merged in https://github.com/necroware/gameport-adapter/pull/48 (which replaced PR https://github.com/necroware/gameport-adapter/pull/45).

https://github.com/necroware/gameport-adapter/tree/sw-ffb-pro

necroware, 2023-02-28:
Unfortunately I still have no FFB Pro joystick. I added correct detection, but there is one report without sufficient information, that the initialization doesn't work. Unfortunately, I can't investigate one that.

Hedda, 2023-03-06:
> Unfortunately I still have no FFB Pro joystick. I added correct detection, but there is one report without sufficient information, that the initialization doesn't work. Unfortunately, I can't investigate one that.

@necroware my offer stands if reconsider a donation of either $30/€30 or one of my Mircosoft SideWinder Force Feedback Pro.

A tip is can still find the original Mircosoft SideWinder Force Feedback Pro for less on different eBay and similar sites in Europe.

* https://www.ebayweb.com/ebay-international-sites
  * http://www.ebay.de
  * http://www.ebay.fr
  * http://www.ebay.it
  * http://www.ebay.com.es
  * http://www.ebay.co.uk
  * http://www.ebay.nl
  * http://www.ebay.ch
  * http://www.ebay.at
  * http://www.ebay.be
* http://allegro.pl
* https://www.ricardo.ch
* https://www.tradera.com

> > @necroware Do you live in the EU and are willing to look at implementing support for MS SideWinder Force Feedback Pro?
> > My OCD made me collect more than a few of the original Microsoft SideWinder Force Feedback Pro joysticks over the years and I might be willing to donate one of my Microsoft SideWinder Force Feedback Pro joysticks to you if you would be willing to then look into adding full force feedback compatibility with it.
> 
> @necroware Maybe it would make more sense for me to ship it or donate/contribute a similar sum to the shipping cost (around $30/€30/£30) to you via PayPal so that you can buy a Microsoft SideWinder Force Feedback Pro joystick from eBay yourself?
> 
> I might perhaps consider donating more if you be willing to send me a few of your Necroware GamePort Adapters in return.
> 
> PS: By the way, I live in Sweden myself.

christian-georg, 2024-10-05:
@necroware If you can walk me through the steps I can look at the output from the serial console.

Hedda, 2025-07-14:
FYI, @NolanNicholson has developed and released the ”Picowinder” project which is a RP2040-based USB adapter for the Microsoft Sidewinder Force Feedback Pro joystick that includes force-feedback communication:

- https://github.com/NolanNicholson/picowinder

In the credits there it sais that the project uses a lot of information from the [adapt-ffb-joy](https://github.com/tloimu/adapt-ffb-joy) project, and from the [reverse-engineering](https://www.descentbb.net/viewtopic.php?t=19061) that went into it. In addition, several forks of adapt-ffb-joy, including those by [juchong](https://github.com/juchong/adapt-ffb-joy) and [jaybee](https://github.com/jaybee-git/adapt-ffb-joy), provided additional insight.