necroware/gameport-adapter #14: Build instructions for both physical device and firmware

nikitalita, 2022-01-01:
Are there instructions available on how to build this device?

And also how to build the firmware?

necroware, 2022-01-01:
You mean, how to solder it? There is not much to say, four resistors, joystick port and an Arduino. And if you want to know how to build an Arduino projects, I'd suggest to start with something simple, like blinking LEDs. The Arduino IDE has a lot of nice examples for the beginners.

creopard, 2022-01-01:
The firmware has to be opened directly in the Arduino IDE. It will then be able to compile it for you and can transfer the compiled result directly to the connected Arduino board via USB.

nikitalita, 2022-01-01:
Where did you get the custom PCB for the dip-24-to-DB15? Can we buy that somewhere?

amak79, 2022-01-03:
@nikitalita

You can use a PCB manufacturing service like JLCPCB or PCBWay. All you need to do is upload the `gameport-adapter-gerber-v0.2.zip` file from the releases page and you will be given a quote.

msgit0, 2022-01-04:
I was hoping for something like a schematic diagram to understand what pins on the DB15 connect to which pins on the Arduino, and how the resistors fit into the circuit.  I've been trying to create a similar device for AGES using a Teensy 2.0 board (similar to Arduino).  My joystick is the Logitech Wingman Digital Extreme, using what I hear is the Logitech ADI protocol (something else I can find no information on).  I have only been successful at reading the state for the basic 4 buttons.

necroware, 2022-01-04:
This is an open source project. Here you can find all the circuits, the PCB, the software and the ADI protocol implementation among other digital protocols. Everything is open, so I really don't get the point of your question.

msgit0, 2022-01-04:
Thank you for the prompt reply.  I have downloaded the code, and it appears I need to install KICAD in order to view the schematic.  I'll do that now.

rdfry, 2022-02-24:
Nikitalita the board is self explanatory.

nikitalita, 2022-02-25:
I built it myself, I understand it now :)

closed 2022-02-25 by necroware