necroware/gameport-adapter #26: FFB wheel giving 0 -512 or 1024 when turning the wheel, only one wheel button works

opensourcerebel, 2022-03-11:
Well more of an info in case anyone hits the same issue.
When I saw this project I was super excited to resurrect my old FFB Sidewinder Wheel.
So what I did was immediately to pull an arduino 23u4 and some Dupoint cables connected to a breadboard and used only the +5V, GND, DATA, CLOCK pins. Device detection was working but package reading was randomly failing. 

The solution was connecting *ALL* of the pins of the MALE DB25 on the sidewinder ffb wheel. I guess some of them are used for reference voltages or the 100K resistors to ground are used for something ... I do not know. 

But in case someone experiences such erratic behaviour with sidewinder FFB wheel - make sure all your adapter pins are connected!

(you can close this issue or may be leave it a bit open for google to index it - however you see fit)

p.s.
For the sidewinder precision pro joystick  +5V, GND, DATA,CLOCK only is sufficient :)

creopard, 2022-06-27:
@opensourcerebel Don't know if that's helpful in any way, but I found some ancient information regarding gameport & wheel support:

> The connection is made so that when both pedals are released, the joystick interface sees about **50 kohm** resistance. When gas is pressed (and break is kept up) the resistance increases and gets maximum resitance when gas pedal is fully pressed down. When the break pedals is fully pressed and gas pedal is released, the pedals system gets the minimun resistance. 

https://web.archive.org/web/20130217101121/http://www.q1.fcen.uba.ar/materias/iqi/joygus/pc_special.html#wheelpedal

and a very nice electrical wiring schematic:
https://web.archive.org/web/20040402124145/http://www.monmouth.com/~lw4750/electrical.htm

necroware, 2023-10-14:
FFB Wheel was reported as working multiple times already, so I close this ticket for now. Should this issue appear again, we can create a new ticket.

closed 2023-10-14 by necroware