necroware/gameport-adapter #29: Support for "WingMan INTERCEPTOR"

gassganso, 2022-04-17:
Interceptor is visually similar to EXTREME with more buttons and hats.
I've tried 1001 configuration (Digital protocol (ADI)) but it shows only one axe an 2 buttons (none of then responfing at all any way).
If is there is anything I can do from my side to help, let me know.

necroware, 2022-04-17:
That joystick should work out of the box. Unfortunately I don't have it to debug, but probably there is some kind of timing issue, so the packet is not decoded properly.

creopard, 2022-06-27:
The Wingman Interceptor has 3 hats in total.
@gassganso did you try the latest version from the main-branch [as the DigitalPin.h was optimized](https://github.com/necroware/gameport-adapter/commit/4e5c81255e7af782aab8c76239a49beb2d331b5b).
Can you also try the debug mode and see what messages the Serial Monitor puts out on initilization?

creopard, 2022-07-03:
As the Interceptor has 3 hats in total, it may be necessary to edit "joystick.h". The maximum of hats are defined as 2:
`MAX_HATS{2u}`

necroware, 2022-07-03:
> As the Interceptor has 3 hats in total, it may be necessary to edit "joystick.h". The maximum of hats are defined as 2: `MAX_HATS{2u}`

Done.

necroware, 2022-09-15:
Any news on this one? Or shall I close it for now?

gassganso, 2022-09-17:
I'll take the project back in the future but it's not possible for me to work on it at the moment. (sorry)

ruenes1, 2022-10-15:
> 

I'm also trying to make a Logitech Interceptor work with the adapter, but so far I've been out of luck. 

I plan to make a dedicated adapter for this joystick and put it inside the base of the joystick, so I have already disassembled it. 
The native gameport cable uses 7 pins, which I've wired to the Arduino as follows:

`Cable color // Gameport pin number // Arduino Pro Micro Pin`
`Black ------------- 4 ------------------------  Gnd`
`Red --------------- 1 ------------------------ VCC`
`Green -------------  14------------------------ D7`
`White -------------  10------------------------ D16`
`Yellow ------------ 7------------------------- D8`
`Orange ------------  2 -------------------------D10`
`Grey --------------   3 ------------------------ A0`

Since there are no analog signals for the Logitech ADI protocol, I've not connected the 100k resitor to pin A0 and gnd, should I?

I see that gameport pins 7, 2 and 3 above correspond to those used in the Logitech.h code, but the joystick also seems to use gameport pins 10 and 14, which I'm not sure are supported by the current code.

When setting the adapter to Logitech mode (which I do by hard coding a "return new Logitech" in the gameport-adapter.ino createJoystick sketch instead of the current code), I get an error when trying to access to the Windows joystick control panel ('your device is not properly connected'). If I manually change the joystick type (e.g., to generic types), the Arduino is properly recognized as the corresponding joystick type (axis/buttons) but, obviously, nothing works. 

On the other hand, if I enable debug messsages in the code, the only info I get on the serial monitor is:

`Detected device: 
`

Do you have any ideas of things I could try? I would be happy to test and debug any change to bring support for this great joystick.

necroware, 2022-10-15:
I didn't test the Interceptor yet, since I simply don't have the device. It should actually work out of the box, there are more pins, than used in the code, because the joystick can be used in different bit modes, as far as I remember. Current software doesn't use that modes and so some signals should remain unused. I guess, there is some issue with packet reading code, may be due to some timing problems. I think some debugging is needed to get an idea, what's wrong.

ruenes1, 2023-02-11:
Just in case it may be of any use, Linux natively supports a variety of Logitech ADI joystick, including the Interceptor. The code for reading data packets with the used parameters and timings can be found here:

https://github.com/exploiteers/Logitech-Revue/blob/master/revue/src/linux/drivers/input/joystick/adi.c

necroware, 2023-02-11:
> Just in case it may be of any use, Linux natively supports a variety of Logitech ADI joystick, including the Interceptor. The code for reading data packets with the used parameters and timings can [...]

Yes it was used as a reference to understand how ADI works. Somebody, who has the Interceptor just has to test it.

Fanbotica, 2023-09-09:
I have a glorious Logitech Wingman Interceptor, I just made a homemade copy of Mr. Necroware's gameport-adapter designed to make the pins available for listening with a Logic Analyzer using dupont cables. I connected the Logic Analyzer like this:

**Logic An.        Arduino         Gameport**
CH1                3                     A0
CH2                2                     10
CH3                7                     8
CH4                10                   16
CH5                12                   5
CH6                14                   7
CH7                15                   9
CH8
GND 
GND               GND               GND

Only the following pins appear to emit signals: the **Gameport 3 pin** (Declared as trigger in the Logitech.h file), the **Gameport 10** "Data 1" pin (which instead in the Logitech.h file appears to be replaced with Gameport 2 "Data 0" pin ), and finally the **Gameport 14** "Data 2" pin (which instead in the Logitech.h file appears to be replaced with Gameport 7 "Data 1" pin).

I proceed to replace the two pins in the firmware that seem incorrect and carry out a series of tests. I hope to update you soon with the results.

Fanbotica, 2023-09-14:
TEST N°1 - I proceeded to set the trigger on 12MIDI_TXD of the gameport in the "logitech.h" file (Following the suggestion of a certain Nuke who says that the only Gameport Output Pin of the game port is 12MIDI_TXD)

FW-name---LOGIC2------LOG-AN-----GAMEPORT-----ARD ProMicro
TRIGGER----Channel4----CH5---------12MIDI_TXD-----5
DATA0------Channel1----CH2----------2BUTTON_1-----A10
DATA1------Channel2----CH3----------7BUTTON_2-----8

IN THIS TEST YOU FINALLY READ THE TRIGGER SIGNAL ON THE CHANNEL OF WHICH I REPORT THE OUTPUT:

HIGH------------LOW
73.718ms-------35.375us
9.458us---------4.002ms
9.458us---------2.001ms
9.458us---------3.001ms
9.458us---------10.003ms
9.458us---------6.002ms
9.458us---------11.003ms
9.417us---------7.002ms
9.417us---------9.003ms
9.458us---------11.057ms
23.625us--------5.467ms
23.625us--------5.221ms
23.667us--------5.221ms...

As regards Channel1 and Channel2 they remain silent, while:

Channel3(GAMEPORT10BUTTON_3/ARD16) and Channel5(GAMEPORT14BUTTON_4/ARD7) emit signals synchronized with some falling edges of Channel4,

Channel7(GAMEPORT13AXYSY_2/ARD-A6) instead repeats the inverted signal of Channel4, spacing the copy signal by 5.083us compared to the original.

Ideas on this?

necroware, 2023-09-14:
MIDI ports on Gameport are usually used only for MIDI and Force Feed Back. Otherwise this neither MIDI RX nor TX should be interesting. Interceptor uses the same protocol as the Extreme, if it doesn't work, may be there is just some timing issues. The code from Logitech.h should work out of the box and if it doesn't I'd analyze if the initialization and reading works properly.

Fanbotica, 2023-09-14:
> Le porte MIDI su Gameport vengono solitamente utilizzate solo per MIDI e Force Feed Back. Altrimenti né il MIDI RX né il TX dovrebbero essere interessanti. Interceptor utilizza lo stesso protocollo di Extreme, se non funziona, potrebbe esserci solo qualche problema di tempistica. Il codice di Logitech.h dovrebbe funzionare immediatamente e, in caso contrario, analizzerei se l'inizializzazione e la lettura funzionano correttamente.

Tnks necroware, at this point I test the default configuration again (logitech.h) and will update you shortly with the times of the signals recorded by the logic analyzer. Then I will subsequently carry out a series of tests on the timing, and for these I will ask for your help.

Fanbotica, 2023-09-14:
These are the test results performed with the default pin configuration (logitech.h). The graphs with the times of the signals recorded by the logic analyzer can be downloaded from here as a "sal" file produced and readable by the Logic 2 software from Saleae Inc:
https://drive.google.com/file/d/1p5_Q8X7FMXgZoZpQYt9xQH2sSPVPB16X/view?usp=sharing

Fanbotica, 2023-09-15:
At the beginning of the trigger sequence before the train of signals interspersed with low levels with the duration: 4, 2, 3, 10, 6, 11, 7, 9, 11 milliseconds, I noticed two high signals, the first of 750ns followed from a low signal of 28ms, then the second of 102us followed by one of 45ms.

Fanbotica, 2023-09-18:
EUREKAAAAAAAAAAAAAAAAAAA!!!!!!! 
Yes!!! Finally!!!! Timing problem solved!!!
After a series of attempts, I managed to find the right duration for the offending timing! This is the duration of the pulse described on line 222 of the "Logitech.h" file which must be modified with "32u", like this:

   Packet readPacket() const {
     static const auto TIMEOUT = 32u;

Thanks to Necroware of course for the magnificent work done but also for the tips provided to resolve the case!

closed 2023-09-20 by necroware