necroware/gameport-adapter #35: Feature Request: Add Schematics

speendo, 2022-09-06:
Hello and first of all thank you for this project!

I tried to revive my old Sidewinder 3D Pro about a year ago utilizing the information from this project (https://www.descentbb.net/viewtopic.php?f=8&t=15526&p=250537#p250537). At least on a modern Linux, I didn't succeed however. The joystick did sometimes show up on the system but the output was more than strange. With this project, I have high hopes to make it work again.

I understand that ordering a PCB and all the other parts is the most convenient solution. However, as I have many "similar" parts at home already, I would like to try and solder the connections myself. Most likely I will be able to figure out which pins need to be connected by looking at the Gerber files.

However, if it was possible to provide the schematics in a separate file (e.g. as an image generated with Fritzing (https://fritzing.org/), this would make it much easier for people like me...

Anyway: thank you again for your effort!

speendo, 2022-09-10:
So I tried to do it myself and came up with this

[image]

[image]

However, I did this for the first time, so I am not sure if it is correct.
@necroware, can you confirm my drawings? In case they are correct I can also provide the Fritzing file and you can of course use them in the repo, if you like.

necroware, 2022-09-10:
Hi, sorry for the late answer, I'm currently short on time. The project contains all the schematics. Just open it with KiCad to see the schematics and the PCB layout. No need to draw it yourself :)

speendo, 2022-09-15:
Thank you!

I feel a little stupid that I didn't find that on myself.

It seems that installing KiCad (which is quite a heavy tool in terms of ressources) was too much of a hurdle to me...

Why not taking a screenshot of the KiCad schematics and adding it to the readme?

necroware, 2022-09-15:
Hi KiCad is not that much heavy and if you work with electronics design, I highly suggest to use something like that. I don't provide an image separately, because I highly dislike to be repetitive and adding an image would mean duplication of what is already included in the KiCad project. Duplication is a bad practice, it always means additional maintenance effort and a source of errors, if s.o. forgets to updater one of the parts. For example image could start to diverge from the KiCad project and result in a lot of confusion. I ran into such problems in another projects already and so I try to avoid duplication as much as possible.

speendo, 2022-09-15:
I understand your argument. Thank you for pointing me the way!

closed 2022-09-15 by necroware