necroware/gameport-adapter #38: Support for additional controllers

crashdot, 2022-09-18:
Hi,

I am wondering if there is anyone that who develops drivers for this adapter that is interested in adding support for additional controllers of this era if I'm willing to donate controllers for this purpose.

Thanks!

CyberLuke-GBG, 2022-09-19:
Hello, I spent last month by adding FFB support in vJoy and FreePIE for USB controllers here: https://github.com/cyberluke/FreePIE/tree/Ffb

I would like to look at Necroware some time, but I don't have Necroware device itself and currently I'm helping two other open-source projects.

Currently I need to learn to compile Windows XP driver using DDK and 7.1 driver. I can compile only kernel Win 10 driver now.

It would be great, if there would be someone, who could do it.

What specific controllers you have in mind? I might reorder my priorities a little bit or it might help others as well.

necroware, 2022-09-19:
@CyberLuke-GBG When talking about drivers, the internal driver in the microcontroller firmware is meant and not Windows/Linux whatsoever.
@crashdot which controllers do you have? May be some of them do work already, or just have to be tested.

crashdot, 2022-09-19:
Greetings,

I may be able to donate a Necroware adapter along with the controllers.

I would need to go through my storage to see if any other controllers are worth considering, but the ones that I can think of immediately are:

Mad Catz Panther XL Joystick (Joystick with trackball)
Interact Force Feedback FX Racing Controller (Radio Controlled car remote style racing controller)
Logitech Wingman Warrior (Joystick with spinner)

The Wingman Warrior has both a gameport and serial port connector, but using the gameport limits its function, since only the joystick part of the controller will work. With the gameport connection, it is supposed to emulate a Logitech Wingman Extreme, however I tried that dip switch setting with the Necroware adapter and it identifies as a "SparkFun Pro Micro" and says "Your game controller is not connected correctly".

With a serial port connection, both the joystick and spinner are available. Someone suggested that the Necroware device could be adapted to translate the serial port input, but I don't know if that is possible. I've tried the Wingman Warrior through a USB to Serial adapter going into DOSbox, but cannot get the entire controller to be recognized. The DOS drivers are intended to install the joystick as a serial mouse and joystick "dual device", but the driver says it only detects one of the devices. In any case, gameport support could be evaluated if serial is not possible.

In any case, I would be willing to donate these and a Necroware adapter if needed (assuming I can find one) to anyone who is willing and able to add support for these controllers.

Cheers

[image]
[image]
[image]

necroware, 2022-09-19:
A driver for the "Interact Force Feedback FX Racing" is already in the works. The "Logitech Wingman Warrior" seems only to be powered through the game port, but the whole communication seems to happen through the serial port, so my adapter is quite useless with it. The "Mad Catz Panther XL Joystick" I see for the first time and probably protocol reverse engineering would be required here.

cyberluke, 2022-09-19:
Yes, we discussed the Interact in another thread, I have that one. I was thinking there be some Windows driver change required, but I guess you can just use sidewinder ffb pid profile as that is what vJoy does on Win XP.

The one with serial port could be worked out using something like FreePIE, manually routing data using Python script. This way I made the Thrustmaster FCS, mark II and weapons systems mark II working. They have AT DIN connector, which I connected to PS/2 adapter and then to USB adapter. The throttle sends real keys 0 to 9 to control throttle and simulate analog control. I just made Python script than converts these keys to joystic axis.

That solution. I’m sharing here: https://gist.github.com/cyberluke/baaadd29bd01fcd6d6e47de3569c3482

In a similar fashion, you could access Arduino or this joystick serial port inside FreePIE. Adding serial port to real Necroware hardware would be the next step, of course. 

I also wrote Teensy USB keyboard to AT DIN, so you can use modern mechanical keyboard with retro design on, lets say 386DX. It would be great if there could be several modules like this in Necroware format and users could just plug n play them using I2C or SPI bus via cable or magnetic pins.

Next thing I’m doing now is connecting two gameport joystics using breadboard, so I can use them on 486 in Descent 1 for 6DOF. The important thing is to use 4 buttons on first joystick, including POV switch and connect only X2,Y2 axis without buttons of second joystick. I will be making Youtube video about that.

I would like to get Necroware and help with Interact code, but don’t see any branch.

necroware, 2022-09-19:
The branch is not yet public, unfortunately I have a very hard deadline on my daily job currently and I don't even have time to release a video on my YT channel.

crashdot, 2022-09-20:
> "Logitech Wingman Warrior" seems only to be powered through the game port, but the whole communication seems to happen through the serial port, so my adapter is quite useless with it. 

The documentation indicates the gameport and serial ports are used separately, and they should not be connected at the same time. If used as a gameport controller, it is supposed to emulate a Logitech Wingman Extreme, although I could not get this working with the Necroware adapter settings for this controller. When connected to the gameport, the spinner does not work. Presumably, gameport support could still be added for this joystick in it's emulation mode.

The serial connector is used on its own, which gives full support for the spinner and joystick. I was able to use an RS232 DB9 to USB adapter to run the Wingman Warrior through DOSBox on Windows 10, where it loads the correct DOS driver and I can play Tempest 2000 for DOS using the spinner and joystick buttons.

This controller can't be used on modern Windows via the serial connector without DOSbox as far as I know, since the drivers are only for DOS or Windows95.

Cheers

necroware, 2022-09-20:
@crashdot I see. But if it emulates "Logitech Wingman Extreme", than you have to set the adapter to "Generic Analog" joystick and not to Logitech. The Wingman Extreme is not a digital joystick, but a generic analog one. You should only use Logitech setting on the adapter with digital joysticks, which use the ADI protocol, like the "Logitech Wingman Extreme Digital".

crashdot, 2022-09-21:
Hi,

Thanks for clarifying that. I can confirm that the Logitech Wingman Warrior seems to work in its two gameport modes with the Necroware adapter as follows:

I don't have a manual for this controller, but I *think* that it is not intended to be able to use the throttle and hat at the same time, but rather there is a switch to select which one you want to use.

1010 - Joystick switch in Hat position = works as 2 axis, 4 button joystick with 4 direction hat controls
1100 - Joystick switch in Throttle position = works as 3 axis, 4 button joystick with the 3rd axis being the throttle wheel

Thanks!

CyberLuke-GBG, 2022-09-21:
@necroware Why it is not public? I remember we are discussing it for over a month or even more. When it will be public? You can tag me here, I will work on it later. But maybe I am busy at that time, who knows....a kid is on the way

crashdot, 2022-09-24:
I tried the different switch settings for the Madcatz Panther XL and could not get any response. It apparently uses a different protocol called A3D. The good news is that someone has apparently figured out the protocol and written Linux drivers for this controller, so perhaps there is documentation that would permit this function to be added to the Necroware adapter at some point. I should be able to donate the controller if needed.

https://github.com/datrh/joyutils

necroware, 2022-09-24:
Yes, the implementation of A3D is available in Linux Kernel directly:
https://github.com/torvalds/linux/blob/master/drivers/input/joystick/a3d.c

It doesn't look too complicated, but I don't have that joystick and currently also very short on time to do the implementation. So may be s.o. else would like to try?

crashdot, 2022-09-24:
I've been storing these controllers for as much as 25 years, so I can surely wait until you have more time to work on such things. Please let me know if you get to a point when you would like me to send the controller to you. Thanks

crashdot, 2023-01-18:
Hi there,
I'm checking to see if there has been any movement on the release of support for the "Interact Force Feedback FX Racing" controller, and if anyone is interested in taking on the "Madcatz Panther XL" as I can donate a controller.

Thanks for your consideration

necroware, 2023-01-18:
Nope, unfortunately still didn't have time for it. Sorry.

beachbouy, 2023-04-12:
I have a Gravis eXterminator gamepad with both serial connector and the USB adapter that came with it. I'd love to get it working again. It last worked on XP with its driver and customising software - Gravis Experience. Is there any chance it would work with the current Gravis "driver" you've implemented? Would it be reasonably easy to add to your code myself? I have some coding experience, but not so much with arduino yet.

Thanks for this awesome project.

creopard, 2023-04-12:
The Gravis Xterminator gamepad seems to send 4 bits long packages (the Gravis Gamepad Pro uses 24 bits, see https://github.com/necroware/gameport-adapter/blob/main/firmware/gameport-adapter/GrIP.h).

In the Linux driver (https://github.com/ilbers/linux/blob/master/drivers/input/joystick/grip.c#L124) exists a function "**grip_xt_read_packet()**" that seems to be a little more complex to initialize the Gravis Xterminator.

Ronnuss, 2024-03-29:
> Interact Force Feedback FX Racing

Did you have any success in implementing the Interact FX Racing Wheel? 
Would be a godsend ;-)

necroware, 2024-03-29:
I started, but unfortunately I had no time to continue.