necroware/gameport-adapter #39: Basic compile problems

AndJ8894, 2022-10-24:
Arduino IDE 2.0.0 gives lots of compile errors. Some header files check for undefined USBCON? 
First two errors:
In file included from C:\github\gameport-adapter\firmware\gameport-adapter\HidJoystick.h:20:0,
                 from C:\github\gameport-adapter\firmware\gameport-adapter\gameport-adapter.ino:18:
C:\github\gameport-adapter\firmware\gameport-adapter\HidDevice.h:20:45: error: expected class-name before '{' token
 class HidDevice : public PluggableUSBModule {
                                             ^
C:\github\gameport-adapter\firmware\gameport-adapter\HidDevice.h:28:25: error: 'HIDSubDescriptor' has not been declared
...

drzacek, 2022-10-24:
Have you chosen the right arduino board in IDE? If you have UNO then it wont work, select arduino leonardo then it has the usb things properly linked and included

AndJ8894, 2022-10-24:
OK, it works! Thanks!