necroware/gameport-adapter #42: Teensy 2.0 support possible

YARD2, 2022-11-28:
Hi,
More a question.
Would it be possible to compile the firmware to work with a Teensy 2.0 ?
BR

YARD2, 2022-12-05:
I'm not very familiar with arduino.
Pin Layout could be changed with compiler directives, I guess.
But I have problems to use the Tennsy extension. It looks like some path are missing during compile.
https://www.pjrc.com/teensy/td_download.html
It also works only with 1.8.19 IDE

YARD2, 2022-12-12:
The main reason for the teensy is, the user friendly FW update Bootloader.
You can just press the button and use a standalone software tool to update the firmware via USB.
I'm not sure if something is available for arduino.

necroware, 2022-12-12:
For Arduino you don't need the studio as well, if you know, what you are doing. I don't use the IDE and have VIM integration. It's just simpler with the IDE for newbies. Eventually it takes longer to write this, than just to take the IDE and flash the MCU ;)

YARD2, 2022-12-12:
Which USB-Bootloader do you use ?

necroware, 2022-12-12:
Arduinos are already delivered with preinstalled boot loader. You usually don't need to install one. But this is out of scope of this ticket. You probably can use Teensy if you want, just adapt the pin IDs in the code `GamePort.h` and flash it to your MCU. I guess, it will work, but as I said, I didn't try.

closed 2022-12-12 by necroware