necroware/gameport-adapter #44: Make SideWinder 3D Pro impersonate SideWinder Precision 2

JimmyGee252, 2023-01-13:
I am wondering if there is a way to make a SideWinder 3D Pro impersonate a SideWinder Precision 2 joystick, 
the joystick USB that effectively replaced the 3D Pro.  By impersonate I mean to push the HID values
that the Precision 2 has.  I own a Precision 2 and while it is useful in that Windows 10/11 and platforms such as Steam recognize the Precision 2 while the 3D Pro is recognized as a generic device in Steam.

The HID information for a Precision Pro shows in Windows 10/11 as:

HID\VID_045E&PID_0038&REV_0105
HID\VID_045E&PID_0038
HID\VID_045E&UP:0001_U:0004
HID_DEVICE_SYSTEM_GAME
HID_DEVICE_UP:0001_U:0004
HID_DEVICE

The HID information for a 3D Pro shows in Windows 10/11 as:

HID\VID_1B4F&PID_9206&REV_0100&MI_02
HID\VID_1B4F&PID_9206&MI_02
HID\VID_1B4F&UP:0001_U:0004
HID_DEVICE_SYSTEM_GAME
HID_DEVICE_UP:0001_U:0004
HID_DEVICE

necroware, 2023-01-14:
What would be the benefit?

creopard, 2023-01-14:
So you would like to rename the "Arduino Micro" entry in the game controller devices list to something else or even to distinguish several connected Necroware gameport adapters?

Like this:
[image]
(The first two entries are separate Necroware gameport adapters, the third entry is actually a real "SideWinder Precision 2" USB-Joystick)

To rename the "Adruino Micro" default name, follow these steps:

1. open the `boards.txt` file in path:
Windows: `C:\Users\<username>\AppData\Local\Arduino15\packages\arduino\hardware\avr\<veraion>\` 
Linux: `/home/<username>/.arduino15/packages/arduino/hardware/avr/<version>/`

2. add the following text at the very bottom(!) of the file and save the changes:
```
micro.build.usb_product="Necroware Gameport Adapter #1"
micro.build.vid=0x2341
micro.build.pid=0x8036
```

3. open the Adruino IDE and flash the connected Arduino Micro as you normally would

4. now the connected device will show up as `Necroware Gameport Adapter #1` in the game controller devices list  (see screenshot above)
If it **still** shows up as "Adruino Micro", open REGEDIT.EXE and delete the key
`HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM\VID_2341&PID_8036`
which will reset the USB device name. (Sorry, I don't know how to reset USB device names in Linux.)

5. If you have several gameport adapter at hand, repeat the above steps for the 2nd, 3rd, etc. gameport adapter. However, make the following changes in the `boards.txt` file, i.e. always increment the ".pid" value by 1 and change the name accordingly:
```
micro.build.usb_product="Necroware Gameport Adapter #2"
micro.build.vid=0x2341
micro.build.pid=0x8037
```

JimmyGee252, 2023-01-14:
> 
The adapter works perfectly for my retro games and some of my newer games as well.  Steam has become the dominate platform for modern gaming and only recognizes the 3D Pro as a “generic game pad” and offers limited use as it tries to map the stick as if it is a game pad.  The Precision 2 is essentially and updated version of the 3D Pro and has a HID recognized by Windows and Steam and is presented as a true joystick, mappable and playable in Steam platform.  So the benefit would be to further extend the usefulness of the gameport adapter and SideWinder 3D Pro.

JimmyGee252, 2023-01-14:
> You would like to rename the "Arduino Micro" entry in the game controller devices list to something else or would even like to distinguish several Necroware gameport adapters connected at once?
> 
> Like this: [image] (The first two entries are separate Necroware gameport adapters, the third entry is actually a real "SideWinder Precision 2" USB-Joystick)
> 
> To rename the "Adruino Micro" default name, follow these steps:
> 
> 1. open the `boards.txt` file in path:
>    Windows: `C:\Users\<username>\AppData\Local\Arduino15\packages\arduino\hardware\avr\<veraion>\`
>    Linux: `/home/<username>/.arduino15/packages/arduino/hardware/avr/<version>/`
> 2. add the following text at the very bottom(!) of the file and save the changes:
> 
> ```
> micro.build.usb_product="Necroware Gameport Adapter #1"
> micro.build.vid=0x2341
> micro.build.pid=0x8036
> ```
> 
>  3. open the Adruino IDE and flash the connected Arduino Micro as you normally would
> 4. now the connected device will show up as `Necroware Gameport Adapter #1` in the game controller devices list  (see screenshot above)
>    If it **still** shows up as "Adruino Micro", open REGEDIT.EXE and delete the key
>    `HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM\VID_2341&PID_8036`
>    which will reset the USB device name. (Sorry, I don't know how to reset USB device names in Linux.)
> 5. If you have several gameport adapter at hand, repeat the above steps for the 2nd, 3rd, etc. gameport adapter. However, make the following changes in the `boards.txt` file, i.e. always increment the ".pid" value by 1 and change the name accordingly:
> 
> ```
> micro.build.usb_product="Necroware Gameport Adapter #2"
> micro.build.vid=0x2341
> micro.build.pid=0x8037
> ```
> 
> @JimmyGee252 since you seem to use another Adruino board, in your case you might need to change `vid=0x2341 pid=0x8037` to `vid=0x1B4F pid=0x9206`

What I am wanting to accomplish is to make the joystick present itself as a newer joystick that is recognized by Windows and the Steam platform as a recognizable device.  Your suggestion is quite interesting and I will look at it closer today once I get time to sit at my OC.  Thank you for the suggestion!

JimmyGee252, 2023-01-17:
>

As my Necroware Gameport Adapter uses the Sparkfun Pro Micro board and not an Arduino Pro Micro, in C:\Users\[userid]\AppData\Local\Arduino15\packages I have Arduino, builtin & Sparkfun folders.  C:\Users\[userid]\AppData\Local\Arduino15\packages\SparkFun\hardware\avr\1.1.13 contains boards.txt also.  I followed your instructions but did not see any changes; the only registry keys still showed Sparkfun.  The Sparkfun Pro Micro section of boards.txt is:

################################## Pro Micro ###################################
#############################################################################
promicro.name=SparkFun Pro Micro

promicro.upload.tool=avrdude
promicro.upload.protocol=avr109
promicro.upload.maximum_size=28672
promicro.upload.maximum_data_size=2560
promicro.upload.speed=57600
promicro.upload.disable_flushing=true
promicro.upload.use_1200bps_touch=true
promicro.upload.wait_for_upload_port=true

promicro.bootloader.tool=avrdude
promicro.bootloader.unlock_bits=0x3F
promicro.bootloader.lock_bits=0x2F
promicro.bootloader.low_fuses=0xFF
promicro.bootloader.high_fuses=0xD8

promicro.build.board=AVR_PROMICRO
promicro.build.core=arduino:arduino
promicro.build.variant=promicro
promicro.build.mcu=atmega32u4
promicro.build.usb_product="SparkFun Pro Micro"
promicro.build.vid=0x1b4f
promicro.build.extra_flags={build.usb_flags}

Suggestions?  We can continue this via email if you like.

JimmyGee252, 2023-01-26:
After working with the adapter and joystick in Steam there is no benefit to making the device impersonate another.  The Steam platform is standardized on the game pad controllers.  Creating mappings within Steam gives functionality to the joystick and adapter.  For these reasons I am closing this issue.  Thank you both for your comments and suggestions!

closed 2023-01-26 by JimmyGee252