necroware/gameport-adapter #46: [SUGGESTION] Please consider running a crowdfunding campaign for a custom Necroware GamePort-Adapter on Crowd Supply or other similar crowdfunding platform?

Hedda, 2023-01-30:
@necroware can I suggest you consider to run a crowdfunding campaign on [Crowd Supply](https://en.wikipedia.org/wiki/Crowd_Supply) for a custom Necroware GamePort-Adapter?

Would love for more people to enjoy and contribute to this project and I think a crowdfunding campaign would help put the word out and grow the user base which at some point would hopefully lead to more actively contributing developers. 

While your gameport-adapter is still a niche product with retro-gaming becoming more popular I think that today and in the near future probably a lot (relativly speaking) of retro-gaming enthusiasts and collectors would be very interested in buying a preassembled and preflashed version of gameport-adapter if it came in a nice enclosure and had the same feature set as your existing gameport-adapter already have.

* Open-source hardware (OSH) licensed (CC BY-SA 4.0?).
* Open-source firmware / open-source software licensed  (GPL-3.0 license?)
* DIP-switches
* Optional: Keeping the PCB "shield" design with Arduino Pro Micro compatibility.

The only things that I think are missing compared to your existing in order to make it a successful project as a crowdfunding campaign:

* Include a custom-designed adapter enclosure/case.
  * Open source the adapter enclosure/case CAD designs to allow users to 3D-print their own case.
    *  See example -> https://www.thingiverse.com/thing:5457006
* Proper use of OSHW logo (with "open source hardware" text) -> https://www.oshwa.org/open-source-hardware-logo/
* Optional: Use custom PCB with [SMD components](https://en.wikipedia.org/wiki/Surface-mount_technology) for [pick-and-place machine](https://en.wikipedia.org/wiki/Pick-and-place_machine) for resistors  all in factory manufacturing?
* Publish versioned precompiled firmware binaries, and optionally provide some kind of easy-to-use firmware flashing tool.
  * Also check out BootloadHID for firmware hex file upgrade via USB -> https://www.obdev.at/products/vusb/bootloadhid.html
* Dedicated community forum, like for example GitHub Discussions in main project repo-> https://github.com/features/discussions
* Documenting the project with wiki -> https://docs.github.com/en/communities/documenting-your-project-with-wikis

Then you should really just need some good marketing by reaching out to different retro-gaming communities to help spread the word.

More advanced ideas (and possibly not achievable given the specifications?):

* Base a new board on the Raspberry Pi Pico RP2040 SoC instead as it would probably get a lot more people interested in the project, see for example the Raspberry Pi Pico Gamepad/Joystick Library -> https://gitlab.com/realrobots/PicoGamepad (which can today be used with https://monsterjoysticks.com/9-pin-joystick-to-usb-adapter )
  * Anyway, fact is RP2040-based products/projects gain more notability simply because Raspberry Pi is so very famous/popular.
* Automatic Controller Mapping with default controller profiles that make the tested and supported joysticks/gamepads work out-of-the-box with RetroArch -> https://www.retroarch.com/index.php?page=controller-autoconfig
  * Perhaps informally partner with Team Libretro to send the prototypes for help with RetroArch compatibility? By the way, they are also working on the development of an unrelated "RetroArch Open Hardware" -> https://www.libretro.com/index.php/introducing-the-retroarch-open-hardware-project/

Maybe even consider asking if someone more, like [Kreeblah](https://github.com/Kreeblah) who designed a similar board based on ATmega32u4RC-MU/ATmega32u4-MU (and he still also maintains updated designs for it to this day), would be interested in partnering on a such project to share the load of running a crowdfunding campaign + getting everything manufactured and shipped?

* https://github.com/Kreeblah/SidewinderToUSBV2
* https://descentbb.net/viewtopic.php?t=15526&sid=fa83df3717719b79c992a53c7489c216&start=287
  * https://descentbb.net/viewtopic.php?t=15526&sid=fa83df3717719b79c992a53c7489c216&start=293

[image]

PS: For some more related inspiration may I recommend you check out Monster Joysticks 9 Pin Joystick to USB Adapter product/project:

* https://monsterjoysticks.com/9-pin-joystick-to-usb-adapter
* https://github.com/MonsterJoysticks/9-Pin-Joystick-to-USB-Adapter
* https://www.youtube.com/watch?v=ljb-X7a2WwU&embeds_euri=https%3A%2F%2Fmonsterjoysticks.com

necroware, 2023-01-31:
Initially, I planned to go in that direction and start an online shop with some of my hardware projects, but unfortunately life plays sometimes a bad joke and currently I have absolutely no time for it. Not even enough time for my youtube channel. Sadly but true.

Hedda, 2023-01-31:
Ah, ok, that is sad to hear. I guess that then also means that you do not have time to arrange for a company like PCBWay (or another similar company) to make it for you and have them sell it from their store so you do not have to carry stock or sell it yourself, or?  ...at least you should be aware to some companies like PCBWay do offer such full shop services where they not only make and assembly your design but also act as a reseller with shipping directly to the customer:

https://www.pcbway.com/project/gifts-sell.html

https://www.pcbway.com/project/gifts.html

Hedda, 2023-02-13:
Recommend moving this to the suggestion section -> https://github.com/necroware/gameport-adapter/discussions