necroware/gameport-adapter #47: How to create an own adapter?

necroware, 2023-01-31:
It is then relatively easy for most hobby/amateur-tinkers to solder the parts by hand as it only has very few simple component, see:

https://www.youtube.com/watch?v=HUmbvfqP784&ab_channel=chadaustinak

## Ordering prototype PCB and assembling yourself

As this is [open-source hardware](https://en.wikipedia.org/wiki/Open-source_hardware) you should know as an option that it is today relatively easy (and inexpensive) to just submit the KiCad PCB layout files directly to some PCB-prototype board manufacturers like Oshpark or PCBWay who will take it then make + ship a few of the PCB boards to you, all without you having to use KiCad yourself or even having to understand the PCB layout:

- https://oshpark.com
   - See example shared project (rev 0.2.1 of the board) -> https://oshpark.com/shared_projects/rzwJjMoP
- https://www.pcbway.com
- https://jlcpcb.com

FYI, Chinese PCB manufacturers will usually make around 20 simple PCB board like this for less than $10 and ship them to you for around $20, so depending on your country's import tax you have those at home for around $30-40 in around a months time.

E.i. just buy the ones from the project BOM (Bill-Of-Matierials) list

https://github.com/necroware/gameport-adapter/blob/main/README.md#bill-of-materials-bom

Note! It is not recommended to buy the Mini-USB variant of Arduino Pro Micro as its header is wider than USB-C and Micro-USB variants), and it is not recommended to buy the Micro-USB variant of Arduino Pro Micro as it is too easy to break the connector.

## Bill of materials (BOM)

The hardware is super simple. To build an adapter you'll need the PCB from this project and following parts:

Part    |  Qty  | [LCSC](https://lcsc.com/) #  | [Digikey](https://www.digikey.com/) #                | [Mouser Electronics](https://www.mouser.com/) # | Comment
--------|-------|---------|--------------------------|----------------------|------------------------------------------
CONN1   |   1   | C77835  | 609-5371-ND              | 523-L77SDA15SA4CH4F  | DB15 female connector
R1..R4  |   4   | C172965 | 13-MFR-25FTE52-100KCT-ND | 603-MFR-25FTE52-100K | 100 kOhm resistors
SW1     |   1   | C15781  | 2449-KG04ET-ND           | 642-DS04T            | DIP-4 switch
U1      |   1   | C72120  | ED3051-5-ND              | 649-DILB24P-223TLF   | DIP24 Socket (optional)
U1      |   1   |   N/A   | 1568-1060-ND             | 474-DEV-12640        | Arduino Pro Micro (ATmega32U4 16MHz, 5V), including two 12 pin header connectors, MicroUSB version (see "Known issues")

Tip! Some other user recommended buying an enclosure case: Hammond Manufacturing 1551RTBU, Digi-Key Part # HM1142-ND.

You can otherwise 3D-print your own case, see example -> https://www.thingiverse.com/thing:5457006

## Firmware

Once you have it and want to flash the firmware yourself (to ensure you have the latest version), you can do that yourself as well.

1. Download source code files from -> https://github.com/necroware/gameport-adapter/tree/main/firmware/gameport-adapter
2. Download the Arduino IDE from -> https://www.arduino.cc/en/software
3. Connect the Adapter with your USB cable.
4. Open "gameport-adapter.ino" included in the source files.
5. In "Tools" -> Port -> select the virtual COM port that says (Arduino Micro).
6. In "Sketch" -> select "Upload" which will compile the firmware and upload it to the Arduino.
7. You are done.

_Originally posted by @Hedda in https://github.com/necroware/gameport-adapter/issues/12#issuecomment-1408782899_

Hedda, 2023-01-31:
> As this is [open-source hardware](https://en.wikipedia.org/wiki/Open-source_hardware) you should know as an option that it is today relatively easy (and inexpensive) to just submit the KiCad PCB [...]

FYI, today a few of those and other PCB manufacturers (including Oshpark and PCBWay) have communities / community website that allows anyone to create and share projects similar to the linked shared project on Oshpark, and doing so can help make it even easier for less experienced people that might be new to that part of the hobby to order their own PCB board directly.

https://oshpark.com/shared_projects
https://oshpark.com/shared_projects/rzwJjMoP

https://www.pcbway.com/blog/help_center/How_to_Share_Projects_in_PCBWay_Community_d27959dc.html
https://www.pcbway.com/project/
https://www.pcbway.com/project/shareproject/

creopard, 2023-01-31:
Also make sure to get the right Arduino board, as they may differ in width:
https://www.youtube.com/watch?v=0losEHACB6A

Hedda, 2023-01-31:
Oh, btw, most open-source hardware licenses allow anyone to make the board to then resell the board and/or product, see ex:

- https://www.tindie.com/products/retrofletch/pc-usb-c-gameport-adapter-sidewinder-retro-joystic/
- https://www.tindie.com/products/retrofletch/pc-usb-gameport-adapter-sidewinder-retro-joystick/

PS: Open-source hardware licenses does however usually not extend to the copyrighted name of the creator or an unique name that the original creator has given to the project/product, so it would normally not be allowed to market and sell it as "Necroware" product, hence a disclaimer is usually used by others to inform that it is not official or sold by the original creator.

Hedda, 2023-01-31:
@necroware You might want to consider enabling the GitHub Discussions for this repository (GitHub collaborative communication forum feature for repositories)  and move this there as it allows for collaboration threads without clogging up the issues section:

https://docs.github.com/en/discussions

https://github.com/features/discussions

Lowe0, 2023-02-09:
I've built a few of these using boards from OSHpark.  They work pretty well, although the resistors can be a bit of a pain.  I think a lead-forming tool will help a lot with that.

Hedda, 2023-02-13:
> ## Ordering prototype PCB and assembling yourself
> As this is [open-source hardware](https://en.wikipedia.org/wiki/Open-source_hardware) you should know as an option that it is today relatively easy [...]

Tip on other one-stop-shop turnkey PCB and embedded circuit board manufacturers partners which offer open project pages:

- Seeed Studio (Seeed Technology) https://www.seeedstudio.com
- Elecrow https://www.elecrow.com
- PCBgogo https://www.pcbgogo.com
- Makerfabs https://www.makerfabs.com
- OSpcb  https://www.cnospcb.com

YARD2, 2023-02-28:
Created a small adapter myself at PCBWay
[image]

necroware, 2023-02-28:
Done.

closed 2023-02-28 by necroware

Hedda, 2023-04-17:
Recommend moving this to the suggestion section under discussions -> https://github.com/necroware/gameport-adapter/discussions