necroware/gameport-adapter #51: Testing Necroware adapter with SwFFBPro

PinkBaron1949, 2023-03-04:
Hi folks, I hope to do this right so here is my current situation:
I now have 5 Necroware boards, 3 with micro connector, 2 with c connector and one SWFFBPro and I did a little bit of testing.
Machines used to test:
ASUS ROG Laptop with Win 10 Home 64 Bit with all updates,
Toshiba Satellite Pro Win 10 Home 32 Bit with all updates,
ASUS Desktop Win 10 Home 64 Bit with all updates.
Used software Arduino v 1.8.19
Firmware v2.1.1 (originally not for SwFFBPro)
Firmware "main" (why is this not v2.1.2 or similar)

1. When flashing all 5 boards with the 2.1.1 firmware and connecting to every machine, the indications are consistent.
2. When connecting, Windows accepts the boards as Arduino pro micro. X,Y Axis do not move and there is no chance to calibrate,
so it seems the Sw is not initilized.
[Sw_Softw_v2_1_1.docx](https://github.com/necroware/gameport-adapter/files/10888935/Sw_Softw_v2_1_1.docx)
3. When connecting it also happens as in the following file documented.
[Sw_Softw_v2_1_1B.docx](https://github.com/necroware/gameport-adapter/files/10888953/Sw_Softw_v2_1_1B.docx)
The X,Y axis is trembling but also reacts at inputs from the joystick, Z axis is trembling but reacts to Z rotation on the joyst., X rotation trembling but reacts to movemant of the X axis, all buttons flicker but correspond to input of the joystick buttons, hats flickers in all direction but also react to coolie hat input on the joystick.
Now I rely on your expertise to investigate the bug. Because I have no knowledge of coding etc. I am only able to test and submit the results.
When using the firmware from the main branch, the results are also consistent with all boards but very different.
[Sw_Softw_main.docx](https://github.com/necroware/gameport-adapter/files/10889021/Sw_Softw_main.docx)
Thats all for today. 
Regards.PB

Hedda, 2023-03-06:
@PinkBaron1949  FYI, for reference, there is another existing issue here -> https://github.com/necroware/gameport-adapter/issues/12 however that is specifically requesting/discussing support for the ["Force Feedback" (haptic technology)](https://en.wikipedia.org/wiki/Haptic_technology) with original [Microsoft SideWinder Force Feedback Pro](https://en.wikipedia.org/wiki/Microsoft_SideWinder#Force_Feedback_Pro) joystick via Necroware gameport adapter/firmware, but I believe that you are reporting issues with the MS SideWindows FFB Pro without "Force Feedback", correct?

PinkBaron1949, 2023-03-07:
Because the existing firmware does not work sofar as to be able to calibrate I can not evaluate the FFb.
But yes the stick is a Sidewinder Force Feedback Pro and works fine under LUFA board. 
As I said, anybody that can give me guidance where to look for either in Windows or elsewhere I will povide the information.
and test.
Complaining about not enough information is easy. I have no knowledge about coding but wil try to get you the support I can, remember there are five boards that are waiting for testing.
H.P.

necroware, 2023-03-09:
You can use the serial console in the Arduino IDE to find out, what's wrong. The serial output can be enabled by commenting the line `#define NDEBUG` in `Utilities.h`. If you are not familiar with serial monitoring, you can enable serial monitor in Tools menu of the Arduino IDE.

It would be also interesting to know, what exactly Windows is complaining about. The Precision Pro uses exactly the same packet format, only the initialization differs a little bit. You can try to comment the lines 109-113 in Sidewinder.h to make the adapter think, that FFB Pro is a Precision Pro. If this works, we have investigate, what it doesn't like about the FFB Pro path.

PinkBaron1949, 2023-03-11:
Nice try but I have no knowledge about commenting the files, please be so kind and tell or do the commenting, then I will do the testing.

necroware, 2023-03-12:
Duplicate #53

closed 2023-03-12 by necroware