necroware/gameport-adapter #52: CH Production joystick support

BlazejosP, 2023-03-07:
Just d'like to announce that I have working version of files for CH  analogue  joysticks on game port which working with their extended buttons.

What is a **CH joystick standard**?

Long time ago in good old DOS days joystick on PC only have 2 axes and 2 buttons. That is why we can play with two joysticks. Then this two Joysticks were merged together and was possible to create joystick with throttle and pedals and four buttons and that was all what was possible to be done on 15 pin joystick port.
CH company try to extended this without loss of compatibility of their joysticks with older DOS games and in 1992 starts to sell **CH Flightstick Pro**. This Joystick start to use simple protocol where some combinations of four button pushed together means a new button. This 4 bit protocol has a 16 possible combination so as you imagine you may have 4 standard buttons + 12 additionally.  Not all combination were used in all their joysticks for example this first **CH Flightstick Pro** only used eight to create 8-way POV HAT. From electronics point of view everyone can create such joystick simple by adding more buttons and using diodes no special  circuit was necessary with any chips. I myself extended my ordinary cheap joystick by adding throttle and 6 buttons. I also build circuit in box which  has pass through for game port additionally buttons and throttle . With this box I have extended  CH Joystick without ordinary with 4 buttons.   Without CH driver such joystick behaviour like normal one in DOS games but since 1992 especially simulators has build in drivers for CH what means that you can use fancy POV HATS and buttons for different task. Also since Win95 you can configure such CH joystick in control panel and use all functionality with Win32 games.

What I found during development?
First I haven't any CH original Joystick or peripherals. Thy were out of reach in DOS times for me.  In 90's i play simulators especially US Navy Fighter  etc. I found in early internet an schematics how extend usual joystick for more buttons and throttle so I start soldering simple circuit in my joystick with diodes  and for my shock  USNF recognise my circuit as a expensive CH joystick. Buttons starts to work throttle even an POV hat. The same I found  in Win95 joystick after configuration also works with all buttons etc. Happy from that success I build with father DIY aileron pedals from Wood and that also works. Then I start to think that with other joystick I can build box with joystick cable pas through and I can have CH joystick with this accessory and ordinary joystick when I connect them without box. So in that case I don't modified second original joystick. Then I stored them for last 20 years . That is how I have now CH like joysticks with throttle and pedals for testing. I also have an late from 2000 Saitek  Cyborg 3D for game port acquired recently which to my surprise also used CH standard to add more buttons. Based on that devices I was able to start thinking with @necroware code for CH Flightstick Pro.

In CH standard they have such joysticks

**CH Flightstick Pro** = 3 Axis 4 standard Buttons + 8 positions HAT -> 4 buttons not implemented
**Combatstick 565F / F-16 Combat Stick** = 3 Axis 4 standard Buttons  + 4 positions HAT + 8 postions HAT -> no buttons  combinations left unused
**Virtual Pilot Pro**  = 3 Axis 4 standard Buttons  + 4 positions HAT + 8 positions HAT -> -> no buttons  combinations left unused
Flight Sim Yolk = -> 3 Axis 4 standard Buttons  + 8 positions HAT + 4 additional buttons

My DIY Joystick -> 4 axes   4 standard Buttons + 4 positions HAT  + 6 additional buttons  -> 2 buttons not implemented (works in DOS and win9x)

So we need for implementation such version of Joystick:

**CH like standard Joystick  *
3 axes 16 buttons

**CH like standard Joystick with pedals *
4 axes 16 buttons

**CH like standard Joystick  4way hat**
3 axes 12 buttons 4 way HAT

**CH like standard Joystick 4way hat with pedals**
4 axes 12 buttons 4 way HAT

**CH like standard Joystick  8way hat**
3 axes 8 buttons 8 way HAT

**CH like standard Joystick 8way hat with pedals**
4 axes 8 buttons 8 way HAT

**CH like standard Joystick  both hats**
3 axes 4 buttons 8 way HAT 4 way HAT

**CH like standard Joystick  both hats with pedals**
4 axes 4 buttons 8 way HAT 4 way HAT

necroware, 2023-03-08:
CH Flightstick (Pro) and CH F16 Combat Flightstick are already implemented. CH joysticks didn't allow to press multiple buttons simultaneously, instead they used those combinations to encode more buttons. If you want to extend this behavior just take a look at the decode function of the named implementations. They are very simple.

t0x1cpain, 2023-03-12:
@BlazejosP i have a working implementation for CH Virtual Pilot Pro, but there are some jitter problems on some of the 6 buttons and the hat switch on the right side of the Yoke, maybe i am doing something wrong. Or maybe my Yoke is faulty i really don't know but maybe ill look in to your code and compare it to my code and ill test it. 
I also have one question  about the CH Virtual Pilot Pro Yoke, what binary number you get when there is no button pressed ?

BlazejosP, 2023-03-12:
When 0000 is chosen so means none button is pressed of 4 button code Button 12 is pressed which working reverse to ordinary buttons. But I commented this button so nothing is working when nothing is pressed.  Check yourself with my code. I have now table where I have all codes for CH joystick with 4-way POV hat and some partially with 8-way HAT but I have no idea how 4-way&8-way together are encoded because I have no such Joystick myself and CH done this in some standard way. If you are interested in that investigation and have necessary hardware I can upload here special code for CH joystick which shows button combinations. Then is only necessary push all buttons and write which of them are encoded in which combination inside table. When all types of combination will be know I think is easy create correct driver even for your fancy **CH Virtual Pilot Pro** because from this what I found code works without any jitter on my Saitek Cyborg 2000  and my diy CH joystick. I'm also suprised that you have some PLA IC in your Virtual pilot because CH joystick buttons where build just with simple diodes to have more buttons :)

BlazejosP, 2023-03-13:
My actually knowledge about CH Joystick will be great if you check with your  **CH Virtual Pilot Pro** how all buttons of 8-way hat are encoded and how is encoded 4-way hat together with 8-way hat? With this knowledge is possible to build in support for all CH Products Joysticks even this DIY like myself I have.

|  Number of Combination| Button 1| Button 2 | Button 3 | Button 4 |4-way HAT joystick|8-way HAT joystick|4-way and 8-way HAT joystick|
| ------------- | ------------- |------------- |------------- |------------- |------------- |------------- |------------- |
| 1 | 0  |0  |0  |0  |Nothing pressed (uncomment unused button 12 to have special reverse button) |Nothing pressed (uncomment unused button ? to have special reverse button) |Nothing pressed (uncomment unused button ? to have special reverse button) |
| 2 | 0  |0  |0  |1  |Button 4 |||
| 3 | 0  |0  |1  |1  |Button 8 |||
| 4 | 0  |1 |1  |1  |Button 7 |||
| 5| 1  |1  |1  |1  |⬆️|⬆️||
|6| 1  |0  |0 |0  |Button 1|||
|7| 1  |1  |0 |0  |Button 6|⬅️||
|8| 1  |1  |1|0  |⬇️|⬇️||
|9| 1  |0  |1|0  |⬅️|↙️||
|10| 1  |0  |1|1  |➡️|↗️||
|11| 0 |1  |0 |0  |Button 2|||
|12| 0 |0  |1 |0  |Button 3|||
|13| 1 |0  |0 |1  |Button 5|↘️||
|14| 0|1  |1 |0  |Button 9|||
|15| 0|1  |0|1  |Button 10|||
|15| 1|1  |0|1  |Button 11|||

t0x1cpain, 2023-03-13:
@BlazejosP Thank you for the detailed information i will check it for sure, i can do more testings after about 2 weeks.
By the way you can see the Yoke from this announcement that i've posted some time ago : https://github.com/necroware/gameport-adapter/issues/43 if you didn't already do. And sure will be in touch 
Cheers

t0x1cpain, 2023-04-28:
@BlazejosP Finally got time to test your code. 
So I did modifications using the code you provide and the results were same. 
Definitely my yoke has hardware failure and maybe there is insufficient power for the USB hub + hums and etc...  so that's the reason for the jitter. Plus I think the encoding is done in different way on this yoke. 
So ill provide my code that works for my yoke and hopefully someone who has fully working same yoke can test the code with it and finalize it. 

Thanks.

BlazejosP, 2023-04-28:
Thanks for checking this on your original hardware @t0x1cpain . I have some idea can you by using original analog joystick 4 buttons check what are the combinations of buttons for your  **CH Virtual Pilot Pro Yoke** with that you can fill the table of buttons and HAT encoding if they are working of course. Even if your yoke are burnt in some way I spouse is easy to repair them with a little bit of soldering skills. Buttons and HAT's on easy diodes circuit and yoke on potentiometer. I can show you how I build DIY imitation CH joystick that way 20 years ago. But before that meaby worth to check this yoke on original DOS computer if you have some around?

t0x1cpain, 2023-04-29:
[image]

Let’s assume the button numberings are like shown on the above picture.
I am using the in-build DEBUG code from Necroware inside the Utilities.h file
Here is the following encoding:
No button pressed – CODE 3 -> 0011

Button 1 pressed – CODE 2 -> 0010
Button 2 pressed – CODE 7 -> 0111
Button 3 pressed – CODE 6 -> 0110 – Have Jitter messes up the Axis
Button 4 pressed – CODE 10 -> 1010 - Have Jitter messes up the Axis
Button 5 pressed – CODE 11 -> 1011
Button 6 pressed – CODE 1 -> 0001

Left HAT UP – CODE 13 -> 1101
Left HAT DOWN – CODE 5 -> 0101
Left HAT LEFT – CODE 15 -> 1111
Left HAT RIGHT – CODE 9 -> 1001

Right HAT UP – CODE 12 -> 1101
Right HAT DOWN – Not working (The physical button is working, but there is a problem with the PLA)
Right HAT LEFT – CODE 0 -> 0000 - Have Jitter messes up the Axis
Right HAT RIGHT – CODE 8 -> 1000 - Have Jitter messes up the Axis