necroware/gameport-adapter #57: Cannot upload code to DEV-12640

Gold-Phoenix, 2023-03-23:
I can compile but not upload the firmware to a DEV-12640, I have tried getting the board on Ardiuno IDE, I've tried using it on Linux, and no matter what I do it just disconnects the DEV-12640 and then says it cannot find it.

Please help as I don't know what to do

creopard, 2023-03-23:
If the IDE won't find the board, try this guide:
https://support.arduino.cc/hc/en-us/articles/4412955149586-If-your-board-does-not-appear-in-the-port-menu

Gold-Phoenix, 2023-03-23:
It shows up on COM but it just disconnects it once i upload via ardiuno
https://imgur.com/a/W0sscNs

necroware, 2023-03-24:
Did you try to upload a simple blink example? Does it work?

Gold-Phoenix, 2023-03-24:
I've just tried this example for my micro controller posted here: 
https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide#example-1-blinkies

It didn't work, it's also worth noting that I've tried both versions of the Arduino IDE. I've also tried it on Arch, Ubuntu, and Windows 10 and nothing worked 

https://imgur.com/a/Vt1KspO

necroware, 2023-03-24:
Then I'm afraid this board is the wrong place to ask this question. Here you will not find a lot of Arduino specialists, nor people who have a lot of experience with the clones, like this one. I think, you would get much more help on appropriate Arduino forum or something similar. May be you can google for some discussion about the module, which you have there?

Gold-Phoenix, 2023-03-24:
So it probably is the micro controller then, and thank you for the suggestion, I've already posted on the sparkfun forum, just waiting on a reply now, thank you for your time, I'll mark this as closed.

closed 2023-03-24 by Gold-Phoenix

Gold-Phoenix, 2023-03-25:
Quick question, what micro controller would you recommend, because you said you dont work with the "clones" but i bought the DEV-12640 from mouser like you said to on the part requirements

Gold-Phoenix, 2023-03-25:
Thank you for the clarification :) it would be my luck if I happened to get a defective one haha. And I can't get it to work so it's probably just a one off thing, I've ordered another one from ebay this time, and it says it was tested before being shipped off so hopefully it works, thank you again :)

creopard, 2023-03-25:
And by the way, please also try another USB(3) cable for the connection - just to make sure.

While using an USB3 adapter, I've come accross some older USB cables (often enclosed with power supplies) that only had the ability to load a battery pack, but won't transfer any data, simply because the data lines (D+, D-) were physically not connected. )

Gold-Phoenix, 2023-03-25:
Thank you for your concern, I ended up trying out about 10 different cables and the only thing that changed was sometimes it just didnt show up in Windows, the results for uploading code were always the same

Gold-Phoenix, 2023-03-31:
Just a little update, this one that i bought off of ebay works no problems: https://www.ebay.co.uk/itm/173204568779?var=471925699230

Had to go into Arch Linux and set some permissions but afterwards the upload worked no issues! so maybe the sparkfun pro micro's arnt a good version to use, thank you again for this project and best of luck to you!