necroware/gameport-adapter #59: Wrong axis read for Throttle on Generic Analog 4 button 3 axis joystick

Retroperator, 2023-04-12:
Necroware thank you for this great project!

I own a Gravis advanced joystick and a Microsoft Sidewinder (analog only early model; see attached pic)
[image]

Both these joysticks have a third axis for throttle so they should work with dipswitch setting 0100 (Generic Analog 4 button 3 axis joystick). But when i select this setting the wrong axis is read for throttle. Only when i select a Generic Analog 4 button 4 axis joystick the throttle is read and show as X rotation.

I've looked at the code but i'm not a programmer so i don't know how to fix it.
Any ideas?

Retroperator, 2023-04-12:
I know that the black wheels are for centering..

The point is that the throttle wheel is on the SAME axis as the Gravis analog joystick that i own and is shown in the video's by Necroware. So it must be some kind of standard..

But even if it isn't and these two joysticks are outliers it must be possible to read the other axis.. I just can't find it in the code.. Any idea's where to look and what to change in the code?

Retroperator, 2023-04-12:
OK i've got it working.. This comment in `AnalogJoystick.h` put me off for a long time! `/// @returns a value between 0 and 255`
It should read `/// @returns a value between 0 and 1023`.

I changed the line `return axis3.get();` to `return 1023-m_axis3.get();`.

Now the throttle is working, on both my joysticks, on the right axis and in the right direction.

Again, i'm not a programmer so i don't know if there is a better solution, but this works perfect for me. Maybe this should be changed in the code, but if my two joysticks are special(they probably aren't) it shouldn't.

necroware, 2023-04-12:
Hi, sorry for the late answer. The wrong comment should be fixed indeed, I'll do that. Regarding the axes, the setting for analog joysticks is generic, it should work for all types of joysticks out there. However, every joystick has own interpretation of axes, on one joystick it's just 3rd dimension, on the other it is throttle, on one it is inverted, on the other not etc. Making it switchable for all models of joysticks out there would mean to install an endless array of switches. So the firmware makes an assumption and assigns the axes in following order X, Y, Z, Rx, Ry, Rz. If your game by accident interprets the Rx axis as throttle, it's fine, another game could interpret Z as throttle as well. So, what you can do is actually reassign the axes and buttons in your operation system settings or the game which you want to play. For example in DosBox you can tell which axis means what, so it will be propagated as a proper CH Flightstick or Thrustmaster to the Dos games inside. I'm using Linux and there I can reassign all axes as well, I guess this works in Windows in Joystick settings too (?)

Retroperator, 2023-04-12:
I'm testing in Windows 10 and it's not possible to change axis and/or invert an axis..
That's why i'm assuming there is some kind of standard for all joystick-axis..
I've not yet tried dosbox as i own a dedicated retropc, with soundcard/gameport, for dos-gaming.. 

I will test the gameport adapter with windows 98 on my retro pc sometime next week.. I'll get back with my findings..

ShoresOfNowhere, 2023-05-28:
Hi!

Just to confirm here that also my 3 axes 4 button joy, which works flawlessly under Windows XP connected to a real game port, showing the third axis as "throttle", does not work properly under Win10 with this adapter, but begins working properly when I swap axis 3 and 4 in AnalogJoystick getAxis function.

As usual, I'm available for testing if needed!

Thanks again Necroware for all your hard work!

Shores

ShoresOfNowhere, 2023-05-29:
I was checking this problem when I came across another aspect that, IMHO, is not optimal.

While the idea of using the starting position of the X-Y axes as the middle position, so that, starting from that position, the axes could simply be re-calibrated every time they are read, is a VERY nice one and works very well, it doesn't work so well for the throttle axis, which in many cases is a continuous axis with no significant middle point.

As it happens on both my joysticks which have a throttle, many of these joys do NOT have any visible marker for the middle position of the throttle, so it's almost impossible to put it in a middle position while connecting the adapter... and thus the calibration is almost always bad, simply because you really did not catch the correct middle point!

@necroware I was thinking of writing an AnalogThrottle addition to the firmware, which would be an analog axis with no meaningful mid point, which would simply adapt to automatic min and max values, mapping them linearly onto the 0 - 1023 range... What would you think of that? :)

This would permit a throttle to be used perfectly even when not centered at startup, just by rolling it fully up and down a few times after connecting the adapter and before starting the game...

In that case, do you think that this should be implemented in the already existing Generic 3 axes 4 buttons joy (going by the idea that all 3 axis joys have the throttle in the third axis), or should a different dip switch config be added keeping the current one unchanged?

Everybody, tell me what do you think of this, thanks!