necroware/gameport-adapter #63: Issues with Logitech Wingman Extreme Digital

bonzaihb, 2023-07-10:
Hi there,

I've just built the adapter to use my old Wingman Extreme Digital with it - unfortunately, it does not want to work properly:
when enabling debug logging and jumpers set to "closed, open, open, closed" (did not have a DIP switch at hand, so I used jumpers instead) I am getting:

Detected device: WMD

so I assume, it is recognized correctly. But it's not doing anything: on W10 it is not recognized at all, and on Linux Mint it's showing as 2 axis, 2 buttons only. Buttons 1+2 are working, everything else no response. The axis behaves really strange as well, jumping all over the place.

Any hint how to check whether its the adapter or the joystick itself? The joystick was left in a corner for sth like 20 years... ;)
My old Gravis 2 Button stick (the black one with 3 buttons and rubber stick) is working with no problems btw...

Cheers,
Christian

necroware, 2023-07-10:
Hi, which firmawre did you use? From main branch, or from particular release? I have one Wingman Extreme Digital at hand, I can check it with the latest version from main once again.

bonzaihb, 2023-07-10:
Thanks necroware, I am using the current main branch.

necroware, 2023-07-21:
Hi, sorry for the late answer. I tested the device and it works fine, but what I noticed going through multiple arduino boards is that the switch is not always detected properly, because I try to read it too fast after the setup. I added some delay there and it should now be more stable. Please try out the latest version from the main branch.

bonzaihb, 2023-07-21:
Hi!
No worries! Just tried it, now it's recognized correctly (most of the time, sometimes takes 2-3 re-connects), buttons working, throttle working, hat working, but the X/Y axis are still messed up. After some waiting time, they seem to normalize in jstest-gtk, but as soon as I just grab the stick, they again start jumping all over the place.
I assume the potis might be dodgy, after all these (25+) years...

Anyways, thanks for your time! :)

necroware, 2023-07-21:
Hm, strange, that it still needs 2-3 re-connects. With all adapters and joysticks which I have it works every time. Before that initialization delay, on some arduinos it also needed some re-connects, but no more. I will keep my eye on it, may be I'll run into it again. Regarding Wingman, yes, the one which I have had also jittery throttle and I had to disassemble the joystick to clean the potis. It worked fine ever since.

necroware, 2023-07-21:
I will close this issue for now. Should it happen again, we can open another one. Btw. this was model independent and the detection could go wrong with any other joystick.

closed 2023-07-21 by necroware

bonzaihb, 2023-07-21:
Quick update: disassembled it and flodded the potis with contact spray, works good now! :)