necroware/gameport-adapter #65: Microsoft Sidewinder Force Feedback Pro: Everything is Jittery and clicking Buttons

sniperwolfpk5, 2023-07-21:
Dear @necroware,

First of all, I appreciate your hard work. Thank you very much for such a brilliant adapter.

> Please note that I have a [**Sidewinder Force Feedback Pro**](https://en.wikipedia.org/wiki/Microsoft_SideWinder#Force_Feedback_Pro) joystick, not a **Sidewinder Precision Pro** joystick. 

> I experimented with the **Arduino Micro Pro** and **Arduino Leonardo** board selections in the Arduino IDE, hoping the issue might be specific to the board type.

I followed the instructions and utilized the available materials to create an adapter using a prototype board. Windows successfully detects and recognizes the joystick, which appears to function correctly as the motors center the joystick upon plugging in the adapter. However, inspecting the Windows joystick manager, I noticed that all the axis movements are jittery and erratic, despite the axes and buttons working. Additionally, some buttons are registering automatic clicks without any input from me. 

**Frame 1**
[image]
**Frame 2**
[image]

In the Arduino IDE, a message is repeated on the serial monitor. `Packet decoding failed 1 time`

[image]

[image]

I suspected the issue might be related to the wiring or connection on the prototype board. To troubleshoot, I decided to redo the connections a second time, starting from scratch. Unfortunately, even after this attempt, the problem persisted, and I couldn't resolve it.

**My thoughts**
- I think the Sidewinder FF Pro may have a different packet, and the decoding function may not work correctly.
- I grounded all the ground points, as shown in the picture below.
[image]

I appreciate your help. Please let me know if you need any more information.

necroware, 2023-07-21:
Hi this looks like timing issue. Try to play with trigger and cooldown duration. I have one FFB here which works fine, but there can be slight differences between revisions.

sniperwolfpk5, 2023-07-21:
I will check and I will update you

sniperwolfpk5, 2023-07-21:
Hi @necroware, I tried to change some of the values in the `cooldown`, `delayMicroseconds(1000)` functions, and the value in `wait_duration` in the `readPacket` function without any luck.

For your information, I am not familiar with coding microcontrollers. I am a software engineer (C#, Asp.net, etc.). It is a bit too much for me unless I understand some basics.

From your videos, I understand you created your function to read digital and analog signals from the game port. Since speed is crucial for this task, you need to adjust delays.

Could you provide more details on the range I should try? You mentioned triggers and what to change in the trigger."

necroware, 2023-07-21:
wait_duration is not, what you should change. What you are interested in is cooldown (1-5 ms) and trigger (5-25 µs). It's hard to analyze without a logic analyzer, but may be you can spot some difference when trying other values. I just tried my FFB Pro and it works flawlessly with the code from the main brunch. What you can try is to disconnect the power supply from the joystick to see, if stuttering disappears or at least changes somehow.

sniperwolfpk5, 2023-07-21:
Thank you for your quick response. I will give it a try and get back to you 🙏

sniperwolfpk5, 2023-07-22:
Hi @necroware, it seems that the release version is outdated. After forking your project, I tested the code from the source, and it's working perfectly, except for some minor issues. The axes appear to be a bit jittery. I'm unsure if the z-axis and Rotation-axis use potentiometers, but I know that the central axis is equipped with an optic sensor. 

I tried different values for the `cooldown` function, ranging from 0 to 6, and I also tried `pulse` values from 18 to 26. However, values below 18 are not working. Do you have any thoughts on the jittery axis issue?

I have recorded a short video for you.
https://youtu.be/N9WF86b-05I

I appreciate your help.

YARD2, 2023-07-22:
Jitter is a common problem with the FFB.
Also in the FFB project some years ago
https://github.com/tloimu/adapt-ffb-joy
some users had this issue.
The only thing what helped was to average the axis which results in a
slow delay.
The problem was the old optical sensor which drifts.

sniperwolfpk5, 2023-07-23:
Well, due to the jittery axis, the joystick is not usable.
@necroware, you were using a joystick tool in your videos. Where can I find that tool?