necroware/gameport-adapter #66: Obtain a own PID and change name

MacGH23, 2023-08-02:
You can obtain an own PID for free at
https://pid.codes/
and add a Profile to the board.txt.
Then you can change the name to e.g.
"Necroware gameport"
instead of "Arduino Leonardo"

creopard, 2023-08-02:
Great idea!

Just posting for reference (boards.txt handling):
https://github.com/necroware/gameport-adapter/issues/44#issuecomment-1382740711

hadess, 2023-09-19:
There's also OpenMoko offering the opportunity to do that:
https://wiki.openmoko.org/wiki/USB_Product_IDs

> As the openmoko products have been discontinued, Openmoko Inc. is opening up the use of its Vendor ID for use by the Free and Open Source software and hardware communities. This means you can simply register your devices and will get an allocation of a Product ID for free.

hadess, 2023-10-19:
I've looked at this, and this would either requires [a lot of manual steps in the Arduino SDK directories](https://www.wronex.com/articles/220211-pro-micro-usb-name/#settings-a-custom-product-name), or building "by hand" without the Arduino IDE.

I've started working on a Makefile but didn't manage to compile successfully before running out of time. I'll look again soon unless somebody beats me to it.