necroware/gameport-adapter #70: Sidewinder 3D Pro does not work in windows but in linux

schuellerf, 2023-10-12:
The discussion started in #69, for me the Sidewinder 3D Pro works in linux but not in windows.
The FFB Wheel I have works in both systems with the same adapter and firmware.

Please hint me how to get proper debug information as I'm more a linux/embedded guy.

The device manager says

```
Device USB\VID_2341&PID_8037&MI_02\8&3153e260&0&0002 had a problem starting.

Driver Name: input.inf
Class Guid: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Service: HidUsb
Lower Filters: 
Upper Filters: 
Problem: 0x0
Problem Status: 0xC00000E5
```

[image]

[image]

schuellerf, 2023-10-12:
More info: My Sidewinder 3D Pro has a serial number starting with 0407XXX and I tested this and the FFB Wheel on two different Windows machines. Both detect the Wheel but not the 3D Pro

creopard, 2023-10-12:
To sum things up:

- Sidewinder 3D does work on Linux, not on Windows _(-> **the Sidewinder 3D does not seem to be broken**)_
- Sidewinder FFB Wheel (same DIP switches) does work on Linux and also works on Windows _(->**the adapter does not seem to be broken**)_
- You already tested on two different Windows machines, but the issue remains.

necroware, 2023-10-12:
Additional conclusion: The joystick works in Linux, but it fails on Windows, that means, that the Sidewinder driver code is working and the USB HID code is probably making problems.

Which firmware did you use? Did you try the latest state from master branch or from an older release?

schuellerf, 2023-10-12:
> Additional conclusion: The joystick works in Linux, but it fails on Windows, that means, that the Sidewinder driver code is working and the USB HID code is probably making problems.
> 
> Which firmware did you use? Did you try the latest state from master branch or from an older release?

Totally agree that it has to do something with USB HID...
I work with "main" branch 😊😉

necroware, 2023-10-12:
Does the adapter work if you switch it, let's say into CH Flight Stick mode? Sidwinder 3D Pro can work in that mode too, but the buttons on the base will not work and the hat would have only 4 ways. If that works you can then look into the init routine of the Sidewinder. For the 3D Pro it has to enable the digital mode and read a packet from the joystick before it can generate the HID descriptor. May be this time is too long for windows and it times out the USB connection before it is able to come back with the registration.

schuellerf, 2023-10-13:
I've now got a logic analyzer, I hope that data helps.
It seems that in linux the magic sequence is run once and then the communication starts (at least to the joystick)
whereas when connecting to windows the sequence seems to be repeated 6 times...
I think the pins are connected correctly but the clock and data2 does not seem to do anything, but that might not affect searching this problem.

Linux (where it works)
[image]

Windows (where the Sidewinder 3d pro is not detected properly)
[image]

Here are the files from PulseView/sigrok if this is of any interest:
[Gameport_Sidewinder_linux.sr.zip](https://github.com/necroware/gameport-adapter/files/12894182/Gameport_Sidewinder_linux.sr.zip)

[Gameport_Sidewinder_windows.sr.zip](https://github.com/necroware/gameport-adapter/files/12894196/Gameport_Sidewinder_windows.sr.zip)

necroware, 2023-10-13:
Hm, that is strange. The driver should work as shown for Linux. It should try to read the data packet and if it fails, trigger the magic sequence to try to set the joystick into a digital mode. Then again reading the data packet which would then tell which Sidewinder joystick it is. In your windows case it looks like it fails to read, then it makes the magic sequence and then fails to read, and again, and again. The joystick seems not to answer for some reason. You can try to play with the functions trigger(), cooldown() and enableDigitalMode(). Try to set the magic duration to other values, s.t. between 115 and 305.

schuellerf, 2023-10-13:
Hmm - I played around with the timings without any success. Then I checked the difference of the FFB Wheel (which works on my windows) and the 3D Pro which is the "Hat" so I just disabled the Hat in the description of the 3d Pro class and now the joystick works in windows (obviously without the hat)
... I'll continue to experiment what the problem might be with Hat and windows...

necroware, 2023-10-14:
Ok, that is an important find. That means, that with the joystick driver everything is ok and we don't need to search in Sidewinder.h. It's just HID USB descriptor which makes problems. The error must be in HidJoystick.h somewhere around Hat registration. Once again it would be nice if you could try the device using CH Flight Stick mode on the adapter and see if that one works.

Meanwhile I'll try to find a windows 10/11 PC somewhere to be able to reproduce it as well. Probably there is some inconsistency in the USB HID descriptor, which is ignored by linux, but is treated more strictly by windows. May be just some padding issues.

schuellerf, 2023-10-14:
Do you have a link to a proper USB/HID specification so I can check myself?

schuellerf, 2023-10-14:
Tested the other modes now:

Position 1 & FlightStick mode: 0010
what works: x,y,rotation,slider,4 buttons,4 direction hat
no buttons at the same time

Position 2 & ThurstMaster mode: 1010
what works: x,y,rotation,4 buttons,4 direction hat
buttons at the same time

as expected!

schuellerf, 2023-10-14:
Some more tests:
    static const Description desc{"MS Sidewinder 3D Pro", 4, 8, 1};   does not work
    static const Description desc{"MS Sidewinder 3D Pro", 4, 8, 0};   works (without hat)
    static const Description desc{"MS Sidewinder 3D Pro", 4, 4, 1};   works (with only 4 buttons)

necroware, 2023-10-14:
Ok that looks more and more like bit padding problem. I do single padding in the end of the sequence and not after each block, may be Windows doesn't like it. What is strange, the latest state has been tested with windows and Sidewinder 3D Pro, but may be various Windows versions and USB drivers can vary in their strictness.

necroware, 2023-10-14:
I changed the way how padding is applied, may be this helps. Unfortunately I didn't find a Windows 10 PC for testing today, so I can't tell if this makes any difference. Please try to run the firmware from the following branch on your Windows PC.
https://github.com/necroware/gameport-adapter/tree/win10-usb-hid-fix

On Linux this code works, but also the previous version was working fine, so I'm not quite sure.

schuellerf, 2023-10-14:
I randomly changed the padding and now it works 🥳
I only tested windows 7 and windows 11 (don't have win10 nearby)

schuellerf, 2023-10-14:
FlightStick mode now does not work but only with the padding 4.
maybe the padding 8 is only necessary for those 8 "hat directions"
i.e. we could change the "hasHat" to "hatDirections" and use 4 or 8 respectively

schuellerf, 2023-10-14:
Sure my last fix is based on your win10-usb-hid-fix branch

schuellerf, 2023-10-14:
I'll try to find some proper documentation of the protocol as I don't like the trial and error approach I just did 😔

closed 2023-12-28 by necroware