necroware/gameport-adapter #76: Support for Quickshot QS-135

schuellerf, 2023-10-20:
Hi!

I really like the project, ordered those PCBs and my kids now have fun using the old Sidewinder joysticks, congratulations on that!

Are there any plans to also support other joysticks with the DE-9 connector?
I have got two Quickshot QS-135 here. I was thinking to extend the firmware which could detect a second PCB layout (by some pin configuration) with hopefully two DE-9 connectors to use two joysticks with _one_ adapter.

I'll have a look if the pro micro has the right (and enough) GPIOs to also cover this usecase.

necroware, 2023-10-20:
I think, there is no real need to add any ports to this adapter. You can just daisy chain a DB9->DB15 joystick adapter and set the switches to generic analog joystick.

necroware, 2023-10-20:
I meant C64 / Amiga / Atari joystick adapters which are already available. They translate the signals into PC compatible gameport signaling. I guess those should just work with my adapter if set to generic analog joystick, but of cours I didn't test any. I just mention them, because there are quite a lot of different joysticks out there and they just need to be translated into PC gameport from where this adapter can take over the translation into USB.

necroware, 2024-02-18:
I guess, this can be closed. For now as proposed we can use a normal DB9->DB15 adapter.

closed 2024-02-18 by necroware