necroware/gameport-adapter #78: Cannot use FFB Wheel in Windows10

Xx-yuta-xX, 2023-12-27:
I created a gameport-adapter and tried to use FFB Wheel on Windows 10 but could not use it.
[image]

[image]

creopard, 2023-12-27:
Please try the firmware files from the current repository instead of the release 2.1.1 (from 2022).
The current repo files contain fixes regarding Sidewinder timings.

Xx-yuta-xX, 2023-12-27:
I tried but could not work.

necroware, 2023-12-27:
This is probably a duplicate of the issue #70 

Please try following branch, we still don't know if that one works.
https://github.com/necroware/gameport-adapter/tree/win10-usb-hid-fix

Xx-yuta-xX, 2023-12-28:
I could fix the problem thank you.

necroware, 2023-12-28:
Open source projects work only if everybody helps. Did the code from that branch help, or did you do something else to get it working? If that branch helped, I will integrate it into the main branch, but I can do it only after confirmation from s.o. who has this issue. It seems, that not everybody who has windows 10 runs into this problem, that's why I need a feedback from s.o. who is affected.

Xx-yuta-xX, 2023-12-28:
The branch code had a bug. Then I rewrite "allign" to align. That is all I did.

necroware, 2023-12-28:
Ok, then I fix that and merge into the main.

closed 2023-12-28 by necroware