necroware/gameport-adapter #85: Sidewinder 3DPro not working with current main116b507

MacGH23, 2024-01-23:
With the current main [116b507]
the Sidewinder 3DPro do not work any longer with my system.
The Source from date: 02 08 2023 is working without problem (except sometimes 1/10 not recognized)
But he current one won't work at all.
I see the device but it can not be startet/initialized as Game device.
Error message in Windows. Cannot start device.

creopard, 2024-01-23:
Confirmed.
Sidewinder 3D works until commit 7ad6c97aee37a20bbb74eaebd898e7ac0855bac8 ("Minor code cleanup") and is broken since a08bcfe907fcf55f333b1bb9cb2e92fbb72fe868 ("Register secondary hats as axes")

I can't see the Sidewinder 3D, it won't show up any more in the Windows game controller settings ("JOY.CPL") .

necroware, 2024-02-17:
Please check the latest state from the main branch.

MacGH23, 2024-02-18:
Just tested.
unfortunately same result. 3D pro ist not recognized corretly

necroware, 2024-02-18:
This should work now. I setup a VM with Windows 10 to be able to test it. Looks like Windows is slightly stricter than Linux on how the USB descriptor has to look like. I could reproduce the issue and now the main branch works for me again.

MacGH23, 2024-02-18:
Thanks. Looks good now !
Make some more tests tomorrow, but it looks that the bug is solved

MacGH23, 2024-02-18:
This also fixes the 3d Pro is not working with the Daisy Chain mod
https://github.com/necroware/gameport-adapter/pull/84

necroware, 2024-02-18:
Perfect! Thank you for testing.

closed 2024-02-18 by necroware