necroware/gameport-adapter #86: General questions

Pcnisse, 2024-01-29:
### Discussed in https://github.com/necroware/gameport-adapter/discussions/55

<div type='discussions-op-text'>

<sup>Originally posted by **necroware** March 13, 2023</sup>
Ask your questions here.</div>

Hi,
First of all thaks for the great work you do👍.
I have this great adapter and it works great with my Logitech WingMan Extreme but when i connect the Logitech WingMan Interceptor it wont recognise it anyone has any sugestions?

Br
Pcnisse

necroware, 2024-01-29:
Unfortunately I have no Interceptor at hand to test, but may be this has a similar problem as #85. I'm currently sick, but I will analyze that in the next days, may be then we know more. Stay tuned.

Pcnisse, 2024-01-30:
Hi,

Thanks for the swift reply. Get well👍

//Pcnisse

ruenes1, 2024-02-01:
@Pcnisse I've just tested the latest sources with my Wingman Interceptor and it works fine. Are you sure you are using the latest source code? The support for the Wingman Interceptor was added recently and it is not included in the releases, which are more than one year old. 

@necroware A small suggestion: I see that the additional hats of the Interceptor are mapped as axes. It may be more standard to map them as additional buttons, as it is the case of modern joysticks with multiple hats.

Pcnisse, 2024-02-01:
@ruenes1  That is probably be the issue, but how do i update with the latest source code, cant seem to find how to.
Any one can point me to directions please?
Br
Pcnisse

ruenes1, 2024-02-01:
Just download it by clicking on the upper-right green button in the git repository labelled 'code' and then choose 'download zip'. You will find the same files as in the releases but they will be up-to-date. Then just re-upload the gameport-adapter sketch to the Arduino.

Pcnisse, 2024-02-01:
Ok, can i use arduino IDE software to update?
//
Pcnisse

ruenes1, 2024-02-03:
Select 'Arduino Leonardo' in the tools->board menu before compiling. Also, select the port of your board within the same menu before uploading it

Pcnisse, 2024-02-03:
Thanks for all the help, updated the board and now the Joystick works great!!!
You are great guys!!!

//Pcnisse