necroware/gameport-adapter #88: question about use and calibration with Sidewinder 3D Pro

tastez, 2024-02-18:
hi all - this possibly is an issue of a broken Sidewinder 3D Pro (and if likely, will appreciate any pointers on threads youve seen for fixing it)

I've got a SW3D Pro that I assume I calibrate with Joy.cpl in windows 11.
Calibration and testing is completely out of whack with the adapter jumpers setup as in the directions.

I am not sure what the autocalibration is on the adapter or at what point it happens but I assume we need to calibrate it in Windows anyway.

https://github.com/necroware/gameport-adapter/assets/78642112/876004cf-49bf-433e-ae86-5cb006625147

The joystick is also listed simply as SparkFun Pro Micro but I assume thats normal.

thanks for any answers!

necroware, 2024-02-18:
For the digital joysticks auto calibration is not happening in the adapter. It shows what it gets from the joystick. I'd suggest to open the joystick and clean it.

tastez, 2024-02-19:
thanks - guess this thing is finally biting the dust. Looks like the resistors, caps or optical sensor/leds are just too old now and adjusting variable resistors or cleaning isnt getting joy.cpl to calibrate it

necroware, 2024-02-19:
If you have a retro PC at hand, you can also try to test it there. Also you can set the adapter into Thrustmaster Analog mode and see if something changes. Also please be sure, that you take the latest state from the main branch. There have been made some adjustments in timing and device registration.

tastez, 2024-02-19:
uploaded the new code but i think i bricked the pro micro. 🤡

closed 2024-02-19 by tastez

necroware, 2024-02-19:
Disconnect the adapter from everything, turn all switches off, reconnect the adapter to USB and try to flash the firmware once again. If nothing works, you can easily unbrick one arduino using another one. They don't even need to be the same.

tastez, 2024-02-19:
all good, 
for other bozos:
i did the GND / RST short twice and got it into bootloader mode then uploaded the sketch to Arduino Micro and i think it works the same as Sparkfun Pro Micro, since sketch seems to run fine.

closing the issue since Im quite sure the Sidewinder is rotten. Will get another one eventually!